nefarius/ScpToolkit #518: ScpToolkit won't detect Controller with Windows on SSD

Xaymar, 2016-11-24:
After installing Windows 10 to an SSD, I have been unable to get my Controller (Gioteck PS3 Wired Controller) to work with ScpToolkit. Windows seems to start too fast for ScpToolkit to actually start up - I don't get any of the startup sounds I used to get when I had Windows installed on a HDD.

Any ideas how to solve this, or how to force a specific driver for the controller?

### Hardware Id stuff
- USB Input Device: USB\VID_25F0&PID_C121&REV_0100
- HID\VID_25F0&PID_C121&REV_0100
- HID\VID_25F0&UP:0001_U:0005
- HID_DEVICE_SYSTEM_GAME
- HID_DEVICE_UP:0001_U:0005

### System:
- Gigabyte H97-D3H-CF
- Intel i5-4690
- Windows 10 64-Bit (AU, Version 1607 Build 14393.447)

### Versions tested:
- https://github.com/nefarius/ScpToolkit/releases/tag/v1.6.238.16010
- https://github.com/nefarius/ScpToolkit/releases/tag/v1.7.277.16103-BETA

Only the BETA version worked on my HDD install, neither of them works on the SSD one.

DanHolli123, 2016-12-08:
not possible for the too fast of a boot as it starts with everything else as it uses drivers and services, the notification part starts with other startup items after you login

the fact you have a SSD versus a HDD makes no difference as the computer uses the same interface for both

if you installed windows 10 yourself it might be missing some drivers for the computer, manually install them from your computer manufacturer's website

try uninstalling the Duel-shock drivers and reinstalling them using the tool included with ScpToolkit

Xaymar, 2016-12-08:
It's a mirrored install, so it matches bit by bit with the HDD, anything that would be missing on the SSD would also be missing on the HDD. It's not too bad though, I can use x360ce for games that don't support the other mode of the controller.

Xaymar, 2017-08-08:
Solved by not using ScpToolkit and instead getting an XInput compatible controller.

closed 2017-08-08 by Xaymar