nefarius/ScpToolkit #521: restart pc after  30 minutes, can't re-pair bluetooth

jang-u, 2016-11-25:
ScpToolkit v1.6.238.16010
PS3 Controller worked on pc via usb or bluetooth
shutdown pc, restart after 30 minutes or more, PS3 Controller worked on pc via usb, but not via bluebooth
if restart pc after several minutes , PS3 Controller worked on pc via bluetooth too

[ScpToolkit_Log-Package_Administrator_1480104170.zip](https://github.com/nefarius/ScpToolkit/files/613834/ScpToolkit_Log-Package_Administrator_1480104170.zip)

jang-u, 2016-11-25:
same problem on motionjoy 
maybe because unsigned driver

EleBathory, 2016-11-28:
Hi,

Try this one and make sure to allow unsigned drivers by your Windows. There are tutotrials on Youtube.
https://github.com/nefarius/ScpToolkit/releases/tag/v1.7.277.16103-BETA

jang-u, 2016-11-30:
ScpToolkit v1.7.277.16103-BETA
gigabyte GC-WB867D-I (intel wireless bluetooth adaptor)
install driver , does not display bluetooth adaptor
PS3 Controller just worked on pc via usb, no wireless

EleBathory, 2016-11-30:
Hi, 
You might have to add the hardware ID of your Bluetooth adapter in the INI file if I'm not mistaken.

jang-u, 2016-12-01:
ScpToolkit v1.7.277.16103-BETA
gigabyte GC-WB867D-I (intel wireless bluetooth adaptor)
edit ScpControl.ini, add gigabyte GC-WB867D-I hardware ID,
install driver, Device 0 intel wireless bluetooth display in uninitialized bluetooth hosts 
initialize all connected devices as bluetooth hosts, bluetooth hosts installed successfully, but Device 0 intel wireless bluetooth does not display in initialized bluetooth hosts , still uninitialized bluetooth hosts 
go next till finish, PS3 Controller just worked on pc via usb, no wireless too

jang-u, 2017-12-17:
Replaced the intel wireless bluetooth adaptor to csr usb bluetooth dongle
ScpToolkit v1.6.238.16010 , worked very well
ScpToolkit v1.7.277.16103-BETA, could not find usb bluetooth dongle
Shortly, ScpToolkit is the best PS3 Controller driver for PC

closed 2018-01-26 by nefarius