nefarius/ScpToolkit #522: PS3 Controller' joysticks too sensitive // Moves by itself

EleBathory, 2016-11-27:
Hi,

PS3 Joysticks are too sensitive. Characters in games where joysticks are supported move by themselves.
When looking at Xinput Tester, LX/LY and RX/RY are moving a little back and forth.

Is there a different way to calibrate joysticks than through XBOX 360 Controller for Windows? If not, can these joysticks be disabled?

Here is the toolkit information that I am using:
[image]

Thank you,

Ybalrid, 2016-11-28:
Have you looked into the "deadzone" settings on the games you want to play? That's generally you can eliminate the problem this way.

EleBathory, 2016-11-28:
Hi,

Thanks for your answer. However, I should have specified that I play fighting games like Melty Blood Actress Again: Current Code and Under Night In-Birth Exe: Late. They're similar to  Street Fighter in many ways. The first 2 games I mentioned do not have deadzone settings nor can I disable joysticks in them.

Thank you,

EleBathory, 2016-11-30:
I found my answer with another GitHub app. See below

For Nefarious app, that would be amazing if Deadzone settings would be available through Xinput Tester or somewhere within the app SCPMonitor or SCPToolkit itself.

Thank you!!

(Answer to my issue)
https://github.com/KrossX/Durazno
Download: http://tiny.cc/magicaltools

On DL page, get this: 
[image]

Follow readme file on first link, and you're good to go.

closed 2016-11-30 by EleBathory