nefarius/ScpToolkit #523: Can't seem to find dualshock 3 in Windows 10

janky1, 2016-11-28:
Hi - installed based on the guide, and when I pull up device manager I see 'PLAYSTATION(R)3 Controller' under 'libusbK USB Devices' whenever then controlled is plugged in. However, I can't seem to find this controller as a recognized device whenever I open an emulator. It is also not recognized when I open 'set up usb controllers'. Also, when I try to re-install the driver, I don't see it come up under either initialized or un-initialized controllers. I feel like I am missing something basic - any help?

Thanks!!

EleBathory, 2016-11-28:
Hi,

Did you download this driver on Microsoft page?
https://www.microsoft.com/accessories/en-us/products/gaming/xbox-360-controller-for-windows/52a-00004#techspecs-connect

Also, I used this version of SCPToolKit and followed all the steps and it worked plugged or with BT.
https://github.com/nefarius/ScpToolkit/releases/tag/v1.7.277.16103-BETA

Before installing anything, uninstall everything, reboot, and lunch the toolkit on GitHub link above. Throughout the install, it will ask you to install the first link above before proceeding. When done, reboot, and they should work and be recognized as Xbox controllers within your games.

I hope this helps.

(Just another user)

DanHolli123, 2016-12-08:
@EleBathory the drivers to get from there are already installed in windows 10

@janky1 have you tried uninstalling all the drivers and reinstalling them?

closed 2018-01-26 by nefarius