nefarius/ScpToolkit #525: Working on USB not Bluetooth

DavoedM, 2016-12-01:
Hello, I have an official DS3 which is working via USB but not via bluetooth.

When on USB I have a playstation 3 controller and an xbox 360 controller showing in devices. On bluetooth none appear. I do get a message saying Genuine Sony Dual Shock 3 found but the controller does not work.

Can you help?

Cheers,
David
[image]
[image]

[ScpToolkit_Log-Package_McColmsPC_1480630003.zip](https://github.com/nefarius/ScpToolkit/files/625856/ScpToolkit_Log-Package_McColmsPC_1480630003.zip)

DanHolli123, 2016-12-08:
you need to install the custom bluetooth driver from the driver installer

DanHolli123, 2016-12-08:
you may also need to reset the controller

DavoedM, 2016-12-08:
Hi,

Thanks for the reply. I've tried a controller reset but it hasn't solved it. 

Where in the driver installer do I find the custom bluetooth driver? Ive been through all the steps, i.e initialising the controller, installing bluetooth hosts, installing the virtual 360 controller driver and the windows service.

There seems to be some bluetooth communication as when I unplug the controller and press the playstation button i get the "Genuine Sony Dual Shock 3 found" message but no controller appears in the devices menu.

Ive just notced another error that appears after installing the drivers and unplugging the controller "SCP Toolkit Warning --HCI_Connection_Complete_EV failed with status 10. Connection handle:0x000D"

Does that mean anything to you?

Cheers,
Dave

binarycore79, 2017-01-12:
Having same issue with DS4. Help, please?

closed 2018-01-26 by nefarius