nefarius/ScpToolkit #526: PS3 Controller not working.

Jazz218, 2016-12-02:
[ScpToolkit_Log-Package_Kostas2_1480679756.zip](https://github.com/nefarius/ScpToolkit/files/627078/ScpToolkit_Log-Package_Kostas2_1480679756.zip)
[image]
I have 2 ps3 dualshock sixaxis controllers and the one is working properly while the other can't.

DanHolli123, 2016-12-17:
Did it work?

Jazz218, 2016-12-19:
Νο, I tried today what you told me but still the one dualshock shows these errors...

Jazz218, 2016-12-26:
So I changed the pcb of the one controller with the pcb of the other and both work now so problem solved!

DanHolli123, 2016-12-26:
Great!
maybe close the issue since you solved your problem
hope you have no more problems either

closed 2016-12-26 by Jazz218