nefarius/ScpToolkit #527: Profile Manager Crash

DanHolli123, 2016-12-02:
This is only a reminder of Issue #252 
I've had this problem and it needs to be readdress as it has been about 8 months and issue #252 was kind of blown off
I want to use the Profile Manager as I want to use the controller as a mouse when outside of games and in some games.
I don't want to go back to the DS3 tool but will if you don't at least TRY to fix it
I DON"T CARE IF IT'S IN ALPHA JUST GET THIS FIXED

nefarius, 2016-12-03:
Very useful comment. Won't change anything though since no one has improved the code since. Nice deadline request, as if you could demand something on a free tool 😆

closed 2016-12-03 by nefarius

DanHolli123, 2016-12-05:
I was just wanting a comment not it to be done by then