nefarius/ScpToolkit #528: Service won't run

Shadow11142, 2016-12-03:
I'm issuing a problem that I'm not sure how to fix. I've installed and uninstalled the scptoolkit multiple times and followed multiple tutorials on installation and ended with the same result. For some reason the drivers install fine, and the dualshock 3 controller is recognized by my pc, but it is unable to read any inputs. When I go into the services menu, The SCP Dsx Service isn't running. When I open the properties and try to run the service it returns an error: "Windows could not start SCP DSx Service service on Local Computer. Error 1068: The dependency service or group failed to start."

Is there any way to resolve this?

nefarius, 2016-12-03:
It means that the `ScpVBus` driver component isn't present or failed to start. Look in your device manager under System Devices for a device with the yellow error mark.

Shadow11142, 2016-12-03:
@nefarius 
You are correct, the ScpVBus driver component has a yellow error mark next to it in my device manager. Is there any way to fix this problem? When I run the SCP Driver installation tool it says that everything was installed successfully, so I'm not sure what went wrong. Thanks for the help!

lehleh, 2016-12-03:
Sorry if I pop up here............. Couldn't work wirelessly, turns out Scp Dsx Service isn't running, then exactly the same as Shadow11142, except that there's no driver marked with yellow mark, nor the driver named "ScpVBus" exists.

nefarius, 2016-12-03:
What's. The. Error. Message. 😞

charlesmilette, 2016-12-03:
Probably that driver signing problem in Win10 AU

nefarius, 2016-12-03:
@charlesmilette yeah, which I could verify if someone would at least think of posting a screenshot so I don't have to rely on my broken crystal ball...

lehleh, 2016-12-03:
Sorry was busy working it around.... Now that I've got the ScpVBus running, but it won't work on bluetooth, it used to work until I reinstalled the application..

Shadow11142, 2016-12-03:
@lehleh 
How exactly did you work around the problem? Right now it's not working for me even over USB :(

lehleh, 2016-12-03:
Because on the other driver installer (the one which required you to tick tick tick checkboxes instead of this guiding wizard), I installed everything by ticking all the boxes and it worked just fine, but with the wizard installer I skipped the "Xbox360" part because I'm using DS3 but when I installed the Xbox360 drivers bunch of drivers installed and it started working.

TL;DR just install everything the Driver Installation offers

nefarius, 2016-12-03:
[I made a video tutorial about the installation](https://www.youtube.com/watch?v=yH0pTK62DAQ&t=2s)

lehleh, 2016-12-03:
I followed the video but at the Bluetooth initializing part my bluetooth device just doesn't show up. Tried uninstalling and reinstalling driver, disabling and reenabling, removing and reinstalling and it still doesn't work.
My hardware ID is this:
Intel(R) Wireless Bluetooth (R)
USB\VID_8087&PID_0A2A&REV_0001

and it's a laptop bluetooth

nefarius, 2016-12-04:
@lehleh you could've saved yourself a lot of time if you'd have [checked out this before](../wiki/Compatible-Bluetooth-Devices).

Shadow11142, 2016-12-04:
@nefarius 
I have figured out to solution to the problem. The root of the problem was that windows couldn't verify a digital signature on the drivers, like @charlesmilette said. The way to fix this is to go into your BIOS and disable secure boot. After doing this, the yellow exclamation point should disappear and your DS3 should be working perfectly. Thank you all for the help :)

closed 2016-12-04 by Shadow11142

charlesmilette, 2016-12-04:
Look into issue #266 for more info on the problem ;)

nefarius, 2016-12-05:
[Well, a problem I could solve](https://www.patreon.com/nefarius) 😉