nefarius/ScpToolkit #531: SCP capturing input from a generic gamepad (dancepad)

JabberwockPL, 2016-12-06:
I have been using SCP with great results for some time now, so I am puzzled with this new problem...

I use SCP to create XBOX virtual controller from FreePIE. Usually it works quite well, except in the following case: I have a cheap dance mat, which reports as „USB Gamepad”. I am trying to remap its input to a virtual XBOX controller. It seems to work, but the XBOX controller _also_ shows the unmapped input. For example, I have assigned button 1 from USB Gamepad as Axis X left in FreePIE. I does work, but XBOX Controller also shows (with interruptions) that button 1 is pressed, even though it is not even assigned in FreePIE. The center of the mat is axis y down in USB Gamepad and it shows as hat down in XBOX controller.

What could be the reason for that?

JohnnyDop3, 2016-12-12:
I have issues using a Gioteck SC-1 controller. It shows up as USB Controller in devices. Seems you figured out how to use ScpServer to make it work. How? :)

JabberwockPL, 2016-12-12:
Are you asking me? Because I have no idea what you are talking about...

closed 2018-01-26 by nefarius