nefarius/ScpToolkit #534: DS4 controller incorrectly reverted to DS3 drivers

Noitanis, 2016-12-06:
After rebooting the computer, or reconnecting the controller the controller won't respond. The ScpMonitor shows that the DS4 controller is interpreted as a DS3 controller, even though DS3 drivers are deselected. Reinstalling the driver seems to be the only thing that help, albeit it's a temporary fix. I don't know if there is any relevance, but once the driver installer prompted that DS3 drivers were installed even though I deselected it before installing.

DanHolli123, 2016-12-08:
try going into the settings and disabling the force Duel-shock 3 drivers and then opening the driver installer, uninstall, deselect Duel-shock 3, reinstall drivers

Noitanis, 2017-01-18:
After following those instructions and running with the setup for little over a month, the problem still occurs. One discovery I did make however is that apparently the controller is interpreted as a DS3 controller in the monitor, but trying to open the settings prompts that the SCP DSx Service is disabled. So the problem might not be that the drivers are reverted as first speculated, but as the service for the drivers gets disabled for some reason. However, I don't know if Ds3Service is the correct service when using a DS4 controller...?

closed 2018-01-26 by nefarius