nefarius/ScpToolkit #536: DS3 works, but doesnt?

bragons, 2016-12-09:
My DS3 has just suddenly started giving me endless problems, sort of. I've used it many many times in the past with SCP and its worked flawlessy up untill recently. I was using an older version which I'd used forever and had no problems, but I recently tried to boot up a game and it just does not register any inputs in the game yet in the monitor it clearly says the controller is working and its picking up all inputs. I know the latest beta steam update had major problems with controllers but I sorted that out. I've wiped all the old drivers off, reinstalled, nothing. Again wiped, updated to a new version, reinstalled 'em, rebooted still nothing. As I said it still picks up inputs in the monitor but thats all. Tried this with Dark Souls and DMC4SE and absolutely nothing, where in the past it did work fine. Any ideas? I'm guessing Windows is breaking something

bragons, 2016-12-09:
Completely a fault on my end, had a steering wheel that was being used as a priority over the controller.

fjverhoek, 2016-12-17:
How did you fix this? Because it does detect my input, but when I try to actually use it inside games it doesnt do anything at all.

closed 2018-01-26 by nefarius