nefarius/ScpToolkit #543: Impossible to start SCP DSx Service

TSlion, 2016-12-15:
Hi there ! 

I am using your excellent software for a while now and since I have my new PC, there is a problem with it.
I was able to install the driver to my controller and play with it. But yesterday, I took my controller and connect it to the PS3 of my brother. When I came back home, I tried to use it but it shows me some errors.

So I decided to install the driver again to the controller but I hade the "Oh Sh... A fatal error occured. Consult the log for details" message. I HATE THIS MESSAGE ^^

I tried some old versions, cleared my PC of all scp stuff (Revo uninstaller, registry...) installed the last version with ScpCleanWipe but still the same problem.

Now, I can see after some long hours of searching that the Scp DSx Service never starting. 
Tried Automatic (delayed), tried everything I saw on different topics in this site but never found a fix. 

I am desesparate man, please help me or I am gonna be crazy !

TSlion, 2016-12-15:
OMG I've found it ! 
The problem came from a stupid thing... 

Right click on "ScpToolkit Driver Installer" **Properties** -> **Compatibility**
Then check "_**Execute this program as compatibility for**_" and then Select "**Vista (Service Pack 2)**"

That's it, all works fine !

DanHolli123, 2016-12-17:
maybe close the issue then?

TSlion, 2016-12-17:
Oh yeah you're right... Thanks !

closed 2016-12-17 by TSlion

jxv16, 2017-11-21:
not works T_T