nefarius/ScpToolkit #544: Programs don't detect PS3 controller after driver installation

jfredi, 2016-12-16:
I'm not the most tech savvy guy in the world so bear with me...

I installed the drivers for my PS3 controller at first with a bluetooth dongle, one built into my computer. It only seemed to work over a wired connection so I was disappointed and tried tinkering with uninstalling and reinstalling the drivers. Now when I connect my controller, it is detected (indicated by the scp notifications and custom LED setup on the controller) but when I try to enter steam Big Picture, it doesn't detect the gamepad. At this point I tried uninstalling and reinstalling the scptoolkit entirely via the setup.exe but when I reinstall, it loads up my previous configuration (not functional) and it's as if I never uninstalled. Any clues as to what I should do? At this point I'd just like to have the controller working via USB.

Any help would be greatly appreciated.

fjverhoek, 2016-12-19:
I have the same problem

jfredi, 2017-01-02:
I found a solution!

What you need to do is first uninstall everything. This includes through the setup program, manually uninstalling the drivers via device manager (ps3 controller, virtual bus, etc.), and manually uninstalling the program itself by going through program files and deleting the Scarlet Crush Productions folder. After this I rebooted my computer, redownloaded the entire program/reinstalled and everything works now. 

I hope this helps you.

closed 2018-01-26 by nefarius