nefarius/ScpToolkit #545: R2 & L2 not responsive on PS3 controler (TB)

hiimjomami, 2016-12-16:
everything works fine thru BT except the R2 and L2 buttons tried the flowing, could use some advice.

-updating BT software
-reinstalling all programs & all drivers
-tried different BT dongle

i dont want to go back to motionjoy plz help, ty so much in advance!

hiimjomami, 2016-12-16:
i guess my biggest question is, is this problem hardware or software related?

DanHolli123, 2016-12-17:
I'm guessing hardware as doing an uninstall of the BT and SIXAXIS (non rumble DS3) drivers and reinstalling with the BT card and the SIXAXIS controller selected it works fine with everything I've thrown at it 
and also my BT card isn't technically supported
maybe test it with MotionJoy and see if the problem continues

hiimjomami, 2016-12-22:
tnx for the response, i am going to try a new dongle since bouth dongles i used where on the old side (V 2.0 ,not shure if tht maters). motionjoy works fine but tru BT, also i am using a DUALSHOCK 3 controler hope tht doesnt make a world of diference.

closed 2018-01-26 by nefarius