nefarius/ScpToolkit #546: Request for compatability with Findway PS3 controllers.

Sweek9, 2016-12-17:
My PS3 controller started bugging out; just old and falling apart internally I guess; so I got a couple Findway Wireless Bluetooth Controllers for PS3 (2 for $17 including shipping). They work perfectly on the PS3, and the PC recognizes it as a "PS(R) Gamepad" but doesnt process inputs (same with a Sony PS3 controller). The Driver Installer seems to install something, then I get a notification about a fake detected and connected to port 1. None of the apps that the genuine controller would work with at this point recognize the inputs though, so I assume these controllers just dont work with SCPToolkit yet.

Sweek9, 2016-12-17:
P.S. I don't have blutooth working, was cabled during the events described.

charlesmilette, 2016-12-17:
It seems that Scp detects it's a fake and tries working around it, but
interpreting it as a genuine would work.

closed 2018-01-26 by nefarius