nefarius/ScpToolkit #549: Can

gustavo-momente, 2016-12-21:


curtiola, 2017-02-25:
I wasn't able to connect my DS3 or DS4 controllers via bluetooth with SCPToolkit alone but now I can have both connected at the same time with a bit of the old ScarletCrush SCP.
here's what I had to do:

I downloaded the SCP-DS-Driver-package-1.2.0.160 from this website:
http://forums.pcsx2.net/Thread-XInput-Wrapper-for-DS3-and-Play-com-USB-Dual-DS2-Controller

once downloaded go into the ../bin/Zadig directory
opened the Zadig program, click device>load preset, load the "bluetooth" preset
then click options>list all devices, select your bluetooth adapter from the list and click "replace driver"
afterwards load the DualShock4/DualShock3 preset (whichever you want to use), go to options>list all devices, choose the controller and click "replace driver"

I hope I'm not too late to help you lol

smsnull, 2017-09-18:
Hi I don't know how late I am to this topic, but I'm having a very similar issue. I've already tried with two different dongles, one with the driver provided by Zadig, the other with the scp driver installer. With both I can get the controller recognized but not working once I disconnect it and try to go by bluetooth. It is a ps3 move nunchuck controller, not the main controller but I've already had it working in my win8 pc, it's the win10 one the one giving me issues.
Have also noticed that the scp pair tool doesn't always displays anything, similarly the SCP monitor either. 
[image]

Kriil, 2017-11-07:
This is probably pretty obvious but just in case (cause this happened to me): Make sure to unplug your PS3/PS4, cause the BT range can be long sometimes and your controller might still be trying to pair with your Playstation.  Couldn't get this working at all until I went downstairs and noticed my Playstation3 was on.  As soon as I unplugged it, problem went away.

closed 2018-01-26 by nefarius