nefarius/ScpToolkit #553: "Scp Virtual Bus Driver" is on a break?

NRG-XLR8R, 2016-12-24:
Win 8 64: in Component Services, I cannot start SCP DSx Service due to its dependency, Scp Virtual Bus Driver. I cannot seem to fathom how to start the Scp Virtual Bus Driver. It is not present in Device Manager.
[image]

hermitmaster, 2016-12-31:
Same thing for me in Windows 10x64. It was working fine for me until I did a clean install of Windows 10. The virtual bus driver appears unsigned to Windows. Not sure how to fix it. I got the driver to successfully install by disabling driver signature checking, but it chokes on reboot.

NRG-XLR8R, 2016-12-31:
My DualShock 3 controller works fine on the Lego Marvel's Avengers game via Steam while plugged in USB, but the above error prohibits Bluetooth (wireless) functionality.

forbjok, 2017-04-08:
I'm having the exact same issue with the 1.7.277 beta driver. I've never had any issues with it on any other PC, but on my new Asus ROG GL502VS laptop, it refuses to load the driver because it's unsigned.
The only relevant difference I can think of is that the laptop came with Windows 10 Home, while all my other PCs run Windows 10 Pro.

closed 2018-01-26 by nefarius