nefarius/ScpToolkit #554: Software breaks webcam/imaging driver

TheFresh16, 2016-12-25:
As mentioned in the title, installing the drivers actually removes the "imaging driver" from my device manager. When pulling up my webcam's camera, it says that it could not be found until after I uninstall the driver. Not a huge deal, but it would be nice to not have to swap back and forth! Thanks!

DanHolli123, 2016-12-26:
have you tried only installing your controller and Bluetooth?

TheFresh16, 2016-12-26:
Yeah I've tried that and various other combinations and the controller only seems to work when I install all of the drivers. Do you think "Force Install" would provide any different results? Thanks for your input!

DanHolli123, 2016-12-27:
I don't think it would, but you can try

DanHolli123, 2016-12-27:
another idea is to uninstall the camera from the scptoolkit in the device manager

closed 2018-01-26 by nefarius