nefarius/ScpToolkit #564: How do I disable the auto updater?

FrozenBeard, 2016-12-30:
Im at the point where Im just tempted to uninstall because the auto-updater pops up every single time I boot my PC, I don't want auto updates, Manual updates are fine, how can I disable this automatic popup at boot? Its infuriating, there needs to be an option to disable this.

closed 2016-12-30 by nefarius

theNailz, 2017-02-03:
https://github.com/nefarius/ScpToolkit/issues/161#issuecomment-172192171

lackofwords, 2018-01-01:
On Windows OS open up your Task Scheduler (under Computer Management if you're still on older OS)

In the "Task Scheduler" folder (expand folder if necessary),
select the "Task Scheduler Library" folder to view the contents.

A small list of task shows in the middle,
Right-click to Disable the one named "updater"

Just incase... Open its properties and view the "Action" tab to ensure it's the Nefarius Software Solutions "ScpUpdater.exe"

Placing the full answer here... Since this is the first thing that popped up in a google search...

ScpToolkit v1.6.238.16010 which is still tagged the latest release, published on Jan 2016. Currently is 107 commits behind master branch, so the "fix" that nefarius has not been applied yet to the latest release.