nefarius/ScpToolkit #578: Problem during driver installation

Quetavian, 2017-01-10:
Hi everyone

When I'm installing the drivers, in the bluetooth initialize step, when I click on Initialize all connected devices as Bluetooth hosts the setup shows this to me: 
[image]
I tried several times, uninstalling scp, skipping the initializing step for bluetooth (but, it doesn't work) and I don't know how to solve this problem.

Quetavian, 2017-01-10:
I cannot upload the log... cause: 
[image]
But I'm trying to upload the zip that the debug creates... I don't know what I'm doing wrong.

PD.- I apologize for my English skills, it's not my first language.

charlesmilette, 2017-01-10:
@nefarius How funny. The error is "The operation completed successfully"

It seems that the installation actually worked, but the error catching in the installer needs improvements

Quetavian, 2017-01-10:
The installation doesn't worked... I can't use the gamepad...

closed 2018-01-26 by nefarius