nefarius/ScpToolkit #580: Fake PS3 controller doesnt connect to laptop

Dienfive, 2017-01-11:
hello guys,

My PS3 controller is from Hopora controller fake: https://hopora.en.alibaba.com/product/60...oller.html

I follow this link https://github.com/nefarius/ScpToolkit to install, however the controller is not recogized in Game controllers (please see attached picture).
I'm using Windows 7, HP 8460p.

Anyone please help me!

Thank you.

[image]

rougetimelord, 2017-01-11:
Read the readme

Dienfive, 2017-01-12:
@rougetimelord Do you mean my controller is not in below list:

Compatible Controllers

To be filled...

USB\VID_054C&PID_0268
Original Sony DualShock 3 Controller
USB\VID_054C&PID_0268&REV_0100
BigBen BB4401 PS3PADRFLX (3rd Party Controller)
USB\VID_0E6F&PID_0214&REV_0580
Afterglow AP.2 Wireless Controller for PS3 (3rd Party Controller)
Although it's a wireless controller, technically it's an USB controller because it uses a proprietary protocol and ships with it's own USB dongle which can't/must not be paired manually.
Rumble, LED-Control and battery charging status isn't supported/implemented yet.

Dienfive, 2017-01-12:
@rougetimelord Do you have other solutions for me?

closed 2018-01-26 by nefarius