nefarius/ScpToolkit #585: Drivers installed sucessfully, ps3 controller shows LED status but no inputs on Xinput tester/games.

squeakyrino, 2017-01-16:
I have a genuine PS3 controller, using usb (No BT).

I installed the drivers of the latest version (v1.7.277.16103-BETA), followed the instructions and everything said it worked fine. I restarted and my PS3 controller worked. After restarting again, it no longer recieves any input from the controller, however it still detects that a controller is plugged in and I can even change the LED status of the controller.

[image]

I do have to note that the battery of my controller is dead.

Xinput tester
[image]

And the logs:
[ScpToolkit_Log-Package_squeakyrino_1484601518.zip](https://github.com/nefarius/ScpToolkit/files/709413/ScpToolkit_Log-Package_squeakyrino_1484601518.zip)

Pav7300, 2017-01-19:
I have the same issue, except I was able to play for about 3 days before this happened...

squeakyrino, 2017-01-20:
Today it worked again. I had the controller unplugged and I hear a pop out of nowhere. It said that the Scarlet Production Service was running. After that I tested and it worked. I have no idea what caused that service to start working now but it seemed to be the issue. Any ideas?

squeakyrino, 2017-01-21:
It seems I have found a fix for this. 

1st. Check your services for the SCP DSx Service, like so:
[image]

2nd. Go to properties and open this tab. 
[image]

3rd. Select "Restart this service" on first failed attempt.

It seems that the first launch of the service is failing and this forces it to restart after it fails. It fixed it for now but it might be worth knowing what's causing the service to fail in the first place.

squeakyrino, 2017-01-21:
Now I have this error:

[image]

closed 2018-01-26 by nefarius