nefarius/ScpToolkit #586: Cant use Bluetooth

DeltaJG, 2017-01-17:
I can use my Playstation 3 Gamepad connected with a cable, through this software, but when i try using it with bluetooth with a bluetooth dongle it connects but doesnt pick up any inputs, I have done all the steps correctly but it still doesnt work,

thaleslipk, 2017-02-26:
Hi i can help you if you have a fake controller i just make a modded ScpControl for 1.7.277 version if you need for some other just let me know but try mine below first.
Works like a charm for 3rd partie controllers.

**[Download the file Click Here!!](http://viid.me/qvnCas)**

If you want instructions just see my tutorial here #652
Bye, im glad if i can help.

DeltaJG, 2017-03-01:
its not a fake controller

closed 2018-01-26 by nefarius