nefarius/ScpToolkit #588: Vibration problem

hitmanboy47, 2017-01-18:
First i wanna say thank you for this software. I used motioninjoy but this is better and more complete.
I have a problem with a game (forbidden siren 2). 
At one point, the game uses the vibration and begin to appear scptoolkit windows. The controller does not work and it becomes disconnected. The problem is solved if i turn off the vibration in the game. Play without vibration is not a problem, however, I wonder if I can resolve this issue. Anyway keep going to improve this tool because is really useful.

hitmanboy47, 2017-01-18:
I am using an original controller dualshock 3.

moogy6610, 2017-01-19:
i have the same problem while using the original controller only , when i use the fake one the vibration works fine

SteamUser, 2017-02-11:
I have the exact same problem with any game that supports vibration. I'm using an original DS3 controller, and I've disabled the rumble through the ScpToolKit, but the problem still exists. If it's on, then I get too many notifications and the controller doesn't get recognized for a few seconds, then works again, then the same again- and again and again. When I turn the rumble off, I get almost the same problem, meaning no vibration or notifications, but the controller doesn't work for those moments when the vibration was supposed to be... I don't know what to do, cause in some games you can disable vibration from their menus, but in some you can't. And it's frustrating.. How can I completely disable any kind of vibration but without getting all that annoying lagging??

SteamUser, 2017-02-14:
After some online searching, I found that this is not a problem caused by ScpToolKit, but the drivers of xbox wired controller itself (???)... Anyway, I think ScpToolKit does need an update, for being able to SAVE a specific setting in the "global configuration" - the one that says "Rumble Latency" under the Dualshock 3 specific settings. For now, I have put that value on the lowest possible (16ms) and it works. No vibration and no lagging at all, no matter if you tick or not the "Disable Rumble" in Global Settings. It would be much better though, if the program would KEEP the 0ms option, cause everytime I put it on 0ms, it automatically changes to to 146ms (or something like that). It should stay at zero, and not change by itself.

SteamUser, 2017-02-14:
When on 0ms it changes to 128ms. Aaand, right now I see that again, it doesn't work as it should... The "disable rumble" option doesn't disable rumble at all, and the "rumble latency" doesn't keep any setting anymore.. Not 0ms, not 16ms, it literally doesn't keep any setting... Wtf is going on? Please fix this....

closed 2018-01-26 by nefarius