nefarius/ScpToolkit #589: Program worked fine yesterday, now my whole PC crashes (blue screen) on start up?!

Rendabomb, 2017-01-18:
Hi,

I downloaded ScpToolkit yesterday to use my Dualshock 3 controller on my Windows 7 (64-bit) pc, and it worked fine. I installed all the drivers, and then chose to have the DS3 installed. As expected it showed up as an X360 controller under devices, got recognized by the emulator I'm using, and I played for a bit using the controller.

Today however, I connected the controller but immediately when the Scp Service booted up and 'connected' with the DS3 my whole PC crashed and I got a blue screen of death. What's going on? I rebooted my pc, now starting the program first before connecting the controller. Same result: crash & blue screen of death.

Then I decided to go for a clean install: I followed the guidelines here on the github and deleted everything, and downloaded the toolkit again. Installing the drivers went fine, but when I chose to install the specific DS3 again my whole pc crashed again the second it was done.

What's going on here? A program like this crashing my whole fucking PC is weird enough, but the fact that it actually already did work yesterday is even weirder. Thanks in advance for any help.

gerciolisz, 2017-01-26:
Hi i have similar problem but in my case first thing is mouse is laggy and when i moving it, it make weird beeping sound. Few times when installing scp drivers again i had BSOD with this : driver verifier detected violation ScpVBus.sys...

[012617-5984-01.dmp.txt](https://github.com/nefarius/ScpToolkit/files/733940/012617-5984-01.dmp.txt)

minidump added just delete .txt

LazorLord, 2017-07-26:
Have the same exact problem.

gerciolisz, 2017-07-26:
If I remember right I made clean install on win 10 and everything was working again.. seems sth broken as I have ds3 using as Xbox pad via bt

closed 2018-01-26 by nefarius