nefarius/ScpToolkit #590: Restart Slow when controller connected

Hyggelik, 2017-01-18:
I am on Win10 Pro x64.
Controller works fine when connected. When I do a restart of the OS, it takes around one minute for the machine to get to the POST. Windows shuts down and I am looking at a black screen for one minute without anything happening. Then I see the POST messages on the screen and the machine restarts.

When I disconnect the controller, the reboot is happening as expected.
Shutting down the computer works correctly, no matter whether the controller is connected or not.

Downloaded on 1/18/17

deadair3210, 2017-01-19:
There is literally no way for the Toolkit to be affecting boot times before Windows is even loaded, more than likely its your computer has something it does with the usb ports thats causing it to wait longer than usual to post

overdrive55, 2017-01-19:
Same here windows 10/64 bit. 
My computer won't post or do anything after a clean install. The ps3 controller worked but I restarted my PC and now nothing happens even if I remove the controller ...

deadair3210, 2017-01-19:
There is literally NO WAY for a user land Windows program with a small driver to interfere with the boot process, as it does not have the permissions necessary even if it wanted to!

Even if there were a gigantic bug in the program that causes Windows to crash as soon as it loads, It STILL wouldnt affect anything before Windows is loaded

charlesmilette, 2017-01-19:
It IS possible for ScpToolkit to crash, hang or affect the computer, because IIRC the Virtual Bus is a kernel driver.

deadair3210, 2017-01-19:
But that would not affect pre-Windows anything, as Windows has almost no say in what happens before it loads unless you are on EFI or UEFI in which case Windows still wouldnt cause anything to fail POST, as the POST test happens before any storage is even looked at for booting

Hyggelik, 2017-01-19:
Thanks for the replies.
Just wanted to clarify:
I would define "boot time" as the time it takes from the BIOS to be initialized, POST and then load up the OS.
That time in my case is NOT affected by whether the controller is plugged in or not.

What is affected is the time between me hitting the "Restart" button in Windows and the time to when the BIOS is initialized. Not sure how to describe it, but I'll give a step by step for the two scenarios:

**1) Controller NOT plugged in: (no increased wait times anywhere)**
- click restart
- windows turns into it's blue "logging off" screen
- black screen, keyboard and mouse are still turned on (I have a backlit keyboard, so I can tell)
- mouse and keyboard turn off
- screen turns back on
- BIOS initialization (where I can hit DEL to get into the setup)
- POST
- Windows is loading
- login etc.

**1) Controller plugged in:**
- click restart
- windows turns into it's blue "logging off" screen
- black screen, keyboard and mouse are still turned on (I have a backlit keyboard, so I can tell)
- **NEW:** Over one minute of wait time, fans are spinning, keyboard is no, but a black screen
- mouse and keyboard turn off
- screen turns back on
- BIOS initialization (where I can hit DEL to get into the setup)
- POST
- Windows is loading
- login etc.

Hope this helps a bit. If you would want me to run a version with some additional debug info in there, I can certainly try.

deadair3210, 2017-01-19:
I think I might have an idea on whats going on. Certain motherboards will take longer if a usb device either refuses ignores or incorrectly responds to a command. Your motherboard might be waiting for the controller to do something for some reason and eventually just gives up and continues after a while. 

In any case, other than something internal to the controller somehow affecting the boot process, there is nothing Windows can do to cause POST to work any differently other than calling a reboot, which causes the POST to go through a less strenuous list of tests

charlesmilette, 2017-01-19:
The Mobo might be trying to boot from the PS3 controller

deadair3210, 2017-01-19:
In that case, it would still be a firmware issue, as, while the controller has some built in flash memory, its only for settings and the controller doesnt export that storage as anything that a computer would be able to boot off of

closed 2018-01-26 by nefarius