nefarius/ScpToolkit #599: Scp Driver overwrites Wifi USB stick under libusbK USB Devices in Device Manager rendering it useless

tecnicalk, 2017-01-24:
This and issue Ive been wrapping my head around to on my lil' brothers computer. When I install SCP driver for his official PS3 controller if I install his Bluetooth driver for Bluetooth card 'CSR8510 A10 (VID: 0A12, PID;0001)' and it overwrite the PS3 controller, Bluetooth card but also the Wifi card for no reason rendering it useless. Any idea on why this happens?

[image]

Also when just installing the PS3 driver on the computer and un-ticking the PS4 controller driver and Bluetooth driver it still installed the PS3 controller driver and the Bluetooth driver over riding it when I check in Device Manage under libusbK USB why does this happen? I guess it must be a bug.

Solopixy, 2017-04-28:
Hi,
I had the same issue. Here's what you need to do
1. Open RUN ( win+R) and type services.msc
2. Look for SCP DSx Service, right click, and select Stop
3. Now you need to open device manager, open RUN (Win+R), type devmgmt.msc
4. Under libusbK USB Devices, right-click on the device you want to fix in this case 802.11 n WLAN and select uninstall
5. A small window will popup, make sure the checkbox is checked for "Delete the driver software for this device" and hit Ok
6. Now on the toolbar at the top, click on the last icon that says "Scan for hardware changes"
and that should do it.

LegendaryDoors, 2017-04-30:
Thank you so much Solopixy, this solved my issue.

closed 2018-01-26 by nefarius