nefarius/ScpToolkit #603: Could you just tell me what to do? Wrong Id?

lovetheories, 2017-01-27:
Short story:

If i have this? **USB\VID_25F0&PID_83C4&REV_0100** . Can't I use it even in wired mode? Why? Can someone which way I could make this fake controller compatible?

I have a DS3 BT controller [http://www.gioteck.com/wp/wp-content/themes/gioteck/images/products/vx2wireless-03.jpg](url) that I would like to use. It has no usb dongle, so it's full ps3. Amazon sent me wrong.

After install devices keeps changing all the time from ps3 to gioteck (endessly...). 

PS: If isnt so hard, tell me how to make it compatible? :/

lovetheories, 2017-02-06:
Nobody?

Wazbat, 2017-02-12:
Did you install a bluetooth dongle with it too?

lovetheories, 2017-05-24:
Hello. Pleaseeeeeeeeeeeeeeeeeee.
Yes I did install it.

closed 2018-01-26 by nefarius