nefarius/ScpToolkit #608: Analog stick moving to the top left.

illusion-games, 2017-01-29:
Hello everyone.

Firstly before I describe my problem, I want to say thank you to Nefarius and everyone who worked on the ScpToolkit as it helped me setup my DS3 controllers without any fuss.

Now my problem is that recently analogs on my DS3 controller somehow got messed up (probably from a little OJ spill) and are pointing to the top left all the time, causing my characters to move and jump.

Now I'm wondering, how to disable analogs because I play only fighting games (MK9, MK X, Injustice etc...) and the characters are always jumping to the left ?

Thanks in forward.

Wazbat, 2017-02-01:
Have you tried just upping the deadzone in the settings manager? If you have it higher than the drift then it should not move.
You can use the windows usb game cotnrolelr test thing to see if it's still drifting I think

illusion-games, 2017-02-01:
Hey, Wazbat.

Thank you for replying, unfortunately even after I put the deadzone to the max the problem is still there because its pointing all the way to the top left corner.

Now I fixed this in x360ce program simply by clearing any assigned buttons on the analogs to amke them disabled, however that program doesnt want to make my L2 and R2 be pressed at the same time so I had to remove it.

Any other suggestions , or if possible if there might be a way to also disable controller buttons ins SCPToolkit ?

Thanks.

closed 2018-01-26 by nefarius