nefarius/ScpToolkit #609: Latest Beta build doesn't detect onboard bluetooth .

masterx1, 2017-01-30:
Hi there,

I have downloaded the latest beta build [ScpToolkit v1.7.277.16103-BETA](https://github.com/nefarius/ScpToolkit/releases/tag/v1.7.277.16103-BETA).

The problem with this build is that the driver installer doesn't detect my onboard bluetooth card. In the stable build it was detect but the problem was that i have to re-install the driver after every system restart/shut down.
@nefarius does this problem have a fix yet ?
[image]

masterx1, 2017-01-31:
Managed to fix it using zadig from previous version of the tool.

closed 2017-01-31 by masterx1

thaleslipk, 2017-02-26:
Hi i can help you if you have a fake controller i just make a modded ScpControl for 1.7.277 version so much better than the older ones if you need for some other just let me know but try mine below first.
Works like a charm for 3rd partie controllers.

**[Download the file Click Here!!](http://viid.me/qvnCas)**

If you want instructions just see my tutorial here #652
Bye, im glad if i can help.

charlesmilette, 2017-02-27:
Can you explain how this could change anything with the driver
installer not detecting his Bluetooth? (which is probably not in the
whitelist of the INI driver file)

Can I also have the source code? If someone just randomly came and
posted a modded DLL in at least 7 threads, my experience is that it's
most of the time a virus.

-- 
Envoyé de mon Gmail

thaleslipk, 2017-02-27:
@charlesmilette No man i just got excited to make my controller connected to bluetooth working perfectly with these modded dll, so actually i read fast i didnt get it his problem was on instalation and i just saw he have to downgrade his version to the old one and i think the latest 1.7 so much better and if his problem was on the 3rd partie controller too he can fix with that dll with no pain because he can still use the latest 1.7 cause i suffer a lot until i got it working uninstaling and installing tak hours cause have to replace the original bluettoh driver you know its painful.
There is no virus, i just wanna help others with my problem or similar maybe he can have a future problem with that so this is a solution right? and i want to spread it out ;), i just wanna help.