nefarius/ScpToolkit #617: PS Button not detected

Random490, 2017-02-06:
I've been trying to get the ps button to work to open up steam while I'm not by my keyboard and mouse but the button isn't working. I've already disabled all the Xbox DVR stuff and when i go to the input detector nothing shows up for the PS button. 
[image]

Wazbat, 2017-02-07:
This is a Microsoft issue, not scptoolkit.
Windows DVR still uses it even when DVR is off. You need to turn it on again, open the game bar, click the setting cog, then untick the "use the Xbox button to open..." box.
Then you can also re-disable dvr

Random490, 2017-02-08:
Didn't realize that you actually had to go into the game bar to turn it off, thank you.

closed 2017-02-08 by Random490