nefarius/ScpToolkit #618: Hardware conflicts when PS4 controller is on and connected

RogueLizard, 2017-02-06:
I noticed a correlation between my Internet connection speed and whether or not my controller is connected through Bluetooth.

When the controller is off and disconnected I get 500KB/s down max (old wireless card)

When controller is ON and CONNECTED I get very low connection speeds. Never goes above 85KB/s. I am very close to the wireless AP.

I have tested this thoroughly under a few different contexts, like having an existing download in progress, using a different wireless adapter, etc.

My specs:
GTX 460 x2 SLI
Intel Xeon X5460 x2
750W 80+ Bronze
2x 160Gb in RAID0
1x Sandisk 240GB SSD
1x Maxtor 500GB 10K RPM

Hardware ID for my Bluetooth:
USB\VID_0A12&PID_0001&REV_1915

Hardware ID for my WiFi card:
PCI\VEN_14E4&DEV_4329&SUBSYS_00601737

My Operating System:
Windows 7 Professional Build 7602

Wazbat, 2017-02-08:
Bluetooth and wifi both transmit on the same frequency, 2.4 GHz, so there is always interference between the two. You could also try with your phone behing held right next to the bluetooth dongle too. 
If you use a 5GHz wifi this shouldnt happen at all. You could test that too.
Also try ethernet.

charlesmilette, 2017-02-08:
His wireless card does not support the 5 GHz band.

Wazbat, 2017-02-08:
I guess you're unlucky then. Maybe try diferent wifi channels? Or just go with ethernet

Wazbat, 2017-02-08:
Or maybe move the dongle to a diferent place on the PC

RogueLizard, 2017-02-08:
I think its because my wifi antennae is 30 db signal and my Bluetooth is 80
db signal. I can't move it, I have no more USB ports. I only have 5.

Wazbat, 2017-02-08:
Have you got any usb extension cables lying around?

RogueLizard, 2017-02-08:
Do I sound rich to you? I have a GTX 460. Of course I don't have any "lying
around", unlike you spoiled rich white people.

Wazbat, 2017-02-08:
"spoiled rich white people" what the fuck"

charlesmilette, 2017-02-08:
Let's try not to start a flame war.

USB extensions cables are quite cheap in stores or online.

Your GTX 460 is still better than my Intel HD 5500.

Wazbat, 2017-02-08:
Where did that come from though. You assume that we're:
Spoiled.
White.
Rich.

We're trying to help you here

RogueLizard, 2017-02-08:
Yeah but it was cheaper ($50USD at time of purchase). I literally don't
have any money. None. I have a dead end job and live in a place that's
basically a hole in the wall. I'd also rather not sell everything I own
just to get a USB cable.

Wazbat, 2017-02-08:
Well don't then. It was just a suggestion. 
Can't you plug your PC into router directly if your house is that small? How are you even affording internet. Or are you getting wi-fi from someplace else. Please explain what your options are

Wazbat, 2017-02-08:
Are you sure it's not your parents paying for your rig? Sell one of your SLI'd GPUs and a xenon to buy food

RogueLizard, 2017-02-08:
I have a one room apartment probably half the size of your kitchen.
Mattress on the floor, desk in the corner. Shared bathrooms and kitchen. I
have a minifridge and microwave and a tiny ass closet.

Am I sure? Yes, I live alone. My parents died when I was 8. I'm not selling
anything to buy anything. I have enough for apartment and food, apartment
comes with poor Internet which is why this is such an issue. Router is in
the ceiling in the hall, 'they' get mad when I touch it. I can't sell my
motorcycle, it's falling apart and worthless, plus I need it to get to work.

Wazbat, 2017-02-08:
I guess in that case your options are to either make your own USB extension cable or just not do any downloads when playing.
See if you can find any old electronic waste. You can get a female USB port from almost any device, then you just need a cable. Then solder/wrap the cables onto the proper female pins. You can find the way it goes on Google.
Or you can get one from a Chinese shop for 1.5$ to 2$
You could also try putting some aluminium foil between the two dongles so it reflects one load of waved away from the other, like the Bluetooth waves forward and let's the WiFi carry on backwards to the router

RogueLizard, 2017-02-08:
I'll try the tin foil, that's one thing I do have lying around

Wazbat, 2017-02-08:
Good luck. You don't want to cover the dongle completely. Imagine that they're both light bulbs and the foil is paper. You want to let the light shine out in a way it reaches your remote, but that it doesn't shine too much on the WiFi bulb

RogueLizard, 2017-02-08:
Okay, I'm not THAT dumb, I'm just unlucky. I built my computer myself
because it was high performance for cheap. Each processor was $35, each GTX
460 was $50 and the board was $40. Toss in a cheap PSU ($35) and some used
HDDS from Goodwill (total of 3 drives for $30), and an SSD from Amazon. My
two Xeons can run circles around a 6700K.

closed 2018-01-26 by nefarius