nefarius/ScpToolkit #619: Phantom input

ZeScratch, 2017-02-08:
So I have a DS3 Genuine (somewhat) controller and the drivers worked perfectly before my controller broke and I basically used it to build a new one from scratch using a friends old one (also DS3 Genuine, same build) as spare parts, so I wouldn't be surprised if I made an error during assembly. However, my Input Tester shows regular results, exactly how it was on the last controller, but my D-pad and buttons were spamming random input, and after a quick clean of the controller again, the bumper buttons are also randomly pressing now and the D-pad is still throwing in random input too.

I trust it isn't an issue with your development, but I would be greatly appreciative of some advice and perhaps if you were to implement an option to make deadzones for all buttons, rather than just joysticks?

Wazbat, 2017-02-08:
I used to have this problem with mine. It's when you're putting it together, I belive it's related to the little ribbon cable that presses onto the other one. Try pressing PS button in the center down hard and tapping it in the center near where that center cable is. It will either make it worse or better. The other option is just try putting it back together again paying special attention to that ribbon cable

ZeScratch, 2017-02-09:
I opened it back up and applied a small amount of rubbing alcohol gently to the contact point of the ribbon cable and the motherboard, which seems to have mostly fixed it. Now I only occasionally have issues with bumpers pressing every few minutes, which is likely due to my frustrated mishandling of it while trying to align it with the frame and the membrane, so I'll have to clean that again when I have the time. Other then that, it might be a little bit of a dodgy board. I didn't own it until a few days ago, so I don't know it's history awfully well.

Still, it would be lovely to have software features that allowed for compensation of minor errors such as that.

Wazbat, 2017-02-09:
Yeah. After pressing it hard on mine it fixed it, but previously they were randomly just slightly pressed. 
+1 for deadzones

closed 2018-01-26 by nefarius