nefarius/ScpToolkit #626: Ghost controller after uninstall.

Sercali, 2017-02-13:
Hey People,

So i sold my PS3 along with the controllers, and i wanted to uninstall SCP since  i wont need it anymore. ( i have a XBO controller now).

Now i've uninstalled Driver ( Force) , uninstalled SCP. but i still see that ''360 controller for windows'' in my devics & printers ( also see it in devmgmt) If i delete it, i'm prompted to reboot and i click yes.

After the reboot i see the ghost controller again in my devices list, it asks if i could reboot, i click no and everything works fine. How do i get rid of this virtual controller completely?

Sercali, 2017-02-13:
Before someone says do the manual uninstall, i followed it. Nothing showed up. The vitual bus only shows up in device manager.

charlesmilette, 2017-02-13:
Remove the virtual bus...

Sercali, 2017-02-13:
Where is it located, How do i remove it?

charlesmilette, 2017-02-14:
In the device manager. Right-click > Uninstall device

Sercali, 2017-02-14:
Read what i said earlier, i already tried that

closed 2018-01-26 by nefarius