nefarius/ScpToolkit #628: Bluetooth issues

nichharp, 2017-02-15:
I've been using the Scptoolkit for a few days now. I've been using my ps3 controller plugged in with no issues whatsoever. However, I recently found one of my old Bluetooth devices and tried to use it. I've been fiddling around with settings and I THINK the drivers are all installed properly however when I pull out the cable my ps3 controller does not connect to the Bluetooth (it does the usual flash all lights many times and then stop). 
I get a popup saying: "Bluetooth host supports communication with DualShock 3 controllers" and another saying "Initializing Bluetooth host 00:15:83:0C:BF:EB (HCI0-Version: 3.0C5C, LMP-Version: 3.0C5C)". So I figured that at least the device is working.
I've tried force uninstalling and force installing the bluetooth but still nothing.
I've tried running the server and not the service. And also tried delaying the start of the service (I read in a forum that worked for people). 
I've tried using other controllers. I've tried reinstalling scptoolkit. And I've tried the bluetooth device on both USB 2.0 and 3.0
I've attached a bug report too but it doesn't mean much to me.
I have no idea what else to do and I'd really appreciate some help.
[ScpToolkit_Log-Package_nichh_1487002276.zip](https://github.com/nefarius/ScpToolkit/files/778083/ScpToolkit_Log-Package_nichh_1487002276.zip)

sollano, 2017-02-16:
I'm having the exact same problem. I even tried updating to beta version, but still can't connect via bluetooth.

Erquint, 2017-02-24:
Bluetooth worked poorly(randomly varying latency) on my knock-off DS3 with the original SCP package. Never worked at all with this fork.

curtiola, 2017-02-25:
I wasn't able to connect my DS3 or DS4 controllers via bluetooth with SCPToolkit alone but now I can have both connected at the same time :)
here's what I had to do:

I downloaded the SCP-DS-Driver_package-1.2.0.160 from this website:
http://forums.pcsx2.net/Thread-XInput-Wrapper-for-DS3-and-Play-com-USB-Dual-DS2-Controller

once downloaded go into the ../bin/Zadig directory
opened the Zadig program, click device>load preset, load the "bluetooth" preset
then click options>list all devices, select your bluetooth adapter from the list and click "replace driver"
afterwards load the DualShock4/DualShock3 preset (whichever you want to use), go to options>list all devices, choose the controller and click "replace driver"

I hope my instructions are clear enough :D lol

nichharp, 2017-02-25:
Thanks for all the attention :)

Curtiola, I tried your method but I'm not seeing any results.
I'll run you through the steps I took and see if I've missed anything.
1)  go to the SCP forum and download
[image]
2) put the file into another folder
[image]
3) Run Zadig, Click "Device" then "Load Preset" and choose the Bluetooth preset
[image]
4) "Options" choose "List all devices" (and make sure it's ticked). Choose the Bluetooth Device and "Replace Driver" [mine says reinstall since I already have it installed]
[image]
5) Repeat steps 3 and 4 but with Dualshock3 preset and the PS3 CONTROLLER.
[image]

I tried running the server and service... nothing. I tried reinstalling and keeping the zadig application open... nothing. 
Should I be using the SCP toolkit at the same time? because I'm not. If, yes, how?

curtiola, 2017-02-25:
Hmm, you did everything the way I did it, for me it worked straight away after I installed the drivers with zadig, I did however use Better DS3 to pair my PS3 controller with my BT adapter earlier in the day though, perhaps pairing is the only issue you're having? (with SCP drivers installed Better DS3 doesn't detect controllers anymore so if you choose to try that, it's a long process with shifty old motioninjoy drivers..)
I honestly don't know how to pair with the SCPToolkit other than during the driver installation.. the DS4 specific settings has an option to "re-pair when connected with USB" but there is nothing like that in DS3 specific settings. 
You could try running the SCPToolkit driver installer again without replacing the BT adapter driver, just do the controller.
Or perhaps your PC is simply blocking unsigned drivers, although I would think it would tell you so if it were, if you haven't disabled the driver signature enforcement, you can do that by opening CMD as administrator [Win + X and select Command Prompt (Admin)] and entering the following:
 `bcdedit.exe /set nointegritychecks on`
I'm sorry if none of this helps, I'm kind of a noob in this department.

& here is what my server looks like
Plugged in:
[image]
Immediately after unplugging it connects with Bluetooth:
[image]

nichharp, 2017-02-26:
Maaaaan, I'm so jealous.
Yh I think the pairing is the real issue here. My SCP doesn't have a dedicated Pair tool.
[image]
My Bluetooth is 4.0
How did you know your Bluetooth was unsupported? (Just trying to find the precise issue)
Version wise, the SCP driver installer is 1.6.238.16010 and the SCPUpdater tells me it's up to date.
This was before trying your Zadig Method. I've tried combinations of using Zadig by itself and with the toolkit, nothing seems to work so far.
Before I'd run the SCPDriverInstaller and if I force install it would freeze and never yeild results (I once left it for over an hour).
I haven't seen any signs of my PC blocking unsigned drivers although I do believe I did this method back when I was using motioninjoy years ago (also couldn't get bluetooth to work). Although i did see this message pop up earlier today and don't really know what it means or if it's even relevant:
[image]

I tried running the ScpDriver.exe but that just gave my controller and bluetooth different drivers and didn't do much:
[image]
[image]
so I changed them back.

My server when plugged in:
[image]
My Server when pulled out:
[image]

Any other ideas? Maybe I'm being a noob and not pairing it correctly?

curtiola, 2017-02-26:
I left the SCPServer & zadig in the downloads folder so it should be fine wherever you have it.
Found out the driver with SCPToolkit is signed, so even if checks are on the drivers would still install..
Also I have no idea what that giant error is about.. but it looks like it's something to do with permissions or it could just be because that DBreezeSchema is already in use, it mentions "creation of database folder failed" perhaps it's as simple as running things as admin? Sometimes it's the smallest problems that take the most time to solve

Here is the list of compatible bluetooth adapters: https://github.com/nefarius/ScpToolkit/wiki/Compatible-Bluetooth-Devices
If you don't know exactly which one you have, you can compare the list with your "USB\VID_0DB0&PID_A871&REV_5276" numbers from the BT adapter in device manager>BT adapter properties
But Zadig should be compatible with any BT 2.0 or higher adapter with EDR as far as I know.
Maybe using the latest beta version of SCPToolkit will work for you, I couldn't say for sure.. but my HCI/LMP version are 4.149c and yours are 3.0c5c (I honestly don't know what that means though)
The process of pairing with Better DS3 after SCPToolkit drivers are installed is long and should be entirely unnecessary.
I'm new to this too lol I have no idea how to pair, as far as I can tell, it should pair automatically when you plug the controller in or when you install the drivers while the controller is plugged in. :/
The beta doesn't have the legacy pair tool either and the one that came in the .zip with zadig in it doesn't seem to work, I just don't plan on ever pairing the controller with my PS3 again.. just in case..

If you want to try it, I don't think it's going to help, but it will guarantee your controller is paired at the very least, here are the instructions to use Better DS3 to pair it:
USE ONLY AS A LAST RESORT

THIS REQUIRES DRIVER SIGNATURE ENFORCEMENT TO BE OFF
[Win + X and select Command Prompt (Admin)] and entering the following:
`bcdedit.exe /set nointegritychecks on`
then
`bcdedit.exe -set TESTSIGNING ON`

then restart your PC

to install the motioninjoy driver for your controller the safest way,
download the driver here: http://www.mediafire.com/download/wjgegko4y8eqyw6/PS3_x64.zip
unzip and put it wherever you want, 
in Control Panel>Hardware and Sound>Devices and Printers
right click your PS3 controller, 
go to properties, 
go to the Hardware tab, 
click "USB Input Device" in the functions list,
click "Properties", 
click "Change settings", 
go to the "Driver" tab, 
click "Update Driver", 
click "Browse my computer for driver software", 
click "let me pick from a list of device drivers on my computer", 
click "Have Disk" 
enter the directory for the driver you've downloaded 
[for me it's "L:\Users\Curtis\Desktop\PS3 x64"] 
and click OK

then use Better DS3 
[ Download here: https://better-ds3.en.uptodown.com/windows ] 
manually set the Master/Host MAC address for the controller using your BT adapters MAC address, which can be found in 
Device Manager > [your BT adapter properties] > details tab > and choose "Bluetooth radio address" from the drop down menu (ignore the last four 0's)
once done, you can start the SCPToolkit driver installation process over from the beginning
& if necessary Zadig as well (try doing the BT adapter without the controller first)
(I only needed Zadig because my BT adapter wasn't supported by SCPToolkit)

thaleslipk, 2017-02-26:
Hi i can help you if you have a fake controller i just make a modded ScpControl for 1.7.277 version if you need for some other just let me know but try mine below first.
Works like a charm for 3rd partie controllers.

**[Download the file Click Here!!](http://viid.me/qvnCas)**

If you want instructions just see my tutorial here #652
Bye, im glad if i can help.

Erquint, 2017-03-03:
@curtiola 
First off, that is kinda offtopic. That is a different project this one is based on.
As far as I can tell, that Zadig process does nothing more than you can do with the built-into Windows Device Manager.
You guys need to learn what drivers and services are.
Services are, basically, GUI-less(invisible) programs that get installed on your Windows that can be run, stopped and restarted. They require no user-input and communicate heavily with other services and programs.
"Launching" a service that is already running does you no good.
All that you need to operate those is already pre-installed in your copy of Windows.
Just type `Services` or `Device Manager` into your Start menu.

nichharp, 2017-03-03:
Erquint,

Yes, we have established that Zadig really didn't do much in the end (but thanks for the help anyway Curtiola). I know what services are. I know that I should either be running the Server OR the Service and not both.
I haven't tried the links that you have sent either, do you think those would help? (considering it didn't work for you) 

P.S. 
I have a genuine PS3 controller so I figured it wouldn't solve my issue.
I had a look at his tutorial and he said "WARNING: THIS FIX MAYBE NOT WORK FOR EVERY FAKE CONTROLLER AND MAYBE CAN BRAKE THE CONNECTIVITY WITH THE GENUINE ONES, YOU HAVE TO TRY AND TEST.
BEFORE YOU DID ANY MODIFICATION PLEASE MAKE A BACKUP OF THE FILES YOU HAVE TO CHANGE." so I wasn't in a rush to try that out.

thaleslipk, 2017-03-03:
@nichharp Man i put that observation that its possible to brake the conectivity with genuines but im not certainly of that because for now my genuine controller is broken and i cant test it, i dont have any genuine ps3 controller to test by now i just have this fake ps3 selled as an original its really similar actually but its not genuine when i test with a genuine controller i will change that and put my results, but if you sure your controller is genuine my fix its really not for you, your problem appears to be maybe you bluetooth dongle.

Erquint, 2017-03-03:
@nichharp 
[image]
The "server" is just a monitor combined with the service.
> I know what services are.
Are you sure? How do you stop the service to avoid running it alongside the "server"?

nichharp, 2017-03-04:
Thaleslipk,
Yeah we said before that it's probably some sort of pairing problem. The bluetooth is supported so it's just annoying. Thanks for your help.
I'll give your method a shot now and get back to you.

Erquint,
I don't appreciate the condescension. I know how to stop the service. I would never be able to run the Server otherwise. I don't, however, know how any of it works and I'm not going to pretend that I do. It's a linking problem. For some reason the bluetooth dongle won't link up to the ps3 controller.

SO, back to the point. Do you think it could be the bluetooth messing up or do you think it's a software issue? The old legacy pairing tool doesn't help whatsoever.

charlesmilette, 2017-03-04:
I've had a similar issue with my Intel AC-3160. The controller would pair with my Bluetooth, but not connect. Might be work checking if the pairing with the computer works (does pressing the PS button starts your PS4?)

nichharp, 2017-03-05:
Thaleslipk,

I tried your method, following the instructions. And I'm not having any luck with it either. The controller won't work wired now.
Your step 3 wasn't very clear "3- Repeat the 2 step until it came to properties of your bluetooth adapter....." What did you mean by that phrase? 

I was getting this error even before i changed ScpControl.ini:
[image]

Other than the possible mistake in step 3, I should have done it fine but I'm leaving screenshots of the changes:
[image]
[image]
[image]
[image]

(Some may say that the pictures are overkill but I'm kinda past the point of caring, not being able to play games wirelessly is kinda killing my mood :P )

thaleslipk, 2017-03-05:
I mean on 3rd step just repeat the step two until you got to the properties
of your device you because the 2nd step is just open this and this to came
to the properties, i said that to cut the same instructions and go directly
to the point but as i see you did this part ok you put the GUID class and
HardwareId correctly.
But my tutorial is just for fake controllers if is not your case the modded
DLL will not help you.
This error that appears looks like you are using maybe a DLL from other
version of ScpToolkit than that one that you have installed, if you are
using the DLL i provide is just for 1.7.277 version of ScpToolkit check if
the DLL you are using is from the same version of the software.

nichharp, 2017-03-07:
Ok, I re-installed the 1.6.238.16010 version. And as it installed, this popped up
[image]
could this be the reason?

thaleslipk, 2017-03-07:
@nichharp Did you already tried to install the 1.7.277 Beta version? [here](https://github.com/nefarius/ScpToolkit/releases/download/v1.7.277.16103-BETA/ScpToolkit_Setup.exe)
obviously using the wipe clean utility to clean all the drivers installed by older versions try that if you cannot make it work.

nichharp, 2017-03-07:
I hadn't used the wipeclean utility before, just regular uninstall through Windows. And yh I did use the Beta Version when I attempted to perform ur method. I will try it again becuase the utility seems to have fixed the issue i had with the beta before.

thaleslipk, 2017-03-08:
When you install a different version you always have to use the wipe clean
utility and then you have to go to your device manager and search for any
drivers left with the "Scp" name, the normal uninstall doesn't clean
everything just the program files, maybe that was the problem for you cause
you probably are running a version of software and another version of
drivers, try again doing all cleanup, good luck!!

thaleslipk, 2017-03-08:
> Still no progress. Wired is working but bluetooth is not.

What version are you running? And maybe your dongle is not supported...

nichharp, 2017-03-14:
Sorry, been away for the weekend.
Ever since upgrading to attempt your method I've been using 1.7.277.16103
And I checked the supported dongles (said somewhere above) and mine was listed.

Masterpoing, 2017-08-05:
Just signed up to GitHub, to say thanks a lot for your help, May God ALWAYS Blessed You!!! Thanks a lot!!!

Kriil, 2017-11-07:
This is probably pretty obvious but just in case (cause this happened to me): Make sure to unplug your PS3/PS4, cause the BT range can be long sometimes and your controller might still be trying to pair with your Playstation.  Couldn't get this working at all until I went downstairs and noticed my Playstation3 was on.  As soon as I unplugged it, problem went away.

EXDarklight, 2017-12-01:
Alright, I'll take a shot at answering this. I saw this somewhere a long time ago and it's what worked for me.
1. Connect up your controller using the cable.
2. Open Zadig and select your bluetooth device. Choose to replace the driver with the "WinUSB" one (I've already done it, so it says reinstall in the image.)
[image]
3. Select the controller and do the same thing
[image]
4. Disconnect cable from controller and check if it connected to bluetooth.
Note that if you use this method, you won't be able to connect any other devices using bluetooth. Only the controller can connect using bluetooth.

closed 2018-01-26 by nefarius