nefarius/ScpToolkit #639: Fake PS3 controller connected PC via USB cable ok but unable to connect via bluetooth

Dienfive, 2017-02-22:
Dear All,

As the title, i've attached picture and log, can someone please take a look? i have other PS3 controllers, connected to PC wirelesslly, but i dont know why it is unable to connect, i've heard someone using other scpcontrol.dll file to fix this, i've tried however still same situation.

when plug usb:
[image]

when unplug USB:
[image]

Log file:

2017-02-21 12:33:25.772 ++ Notify [Arrival] [{E2824A09-DBAA-4407-85CA-C8E8FF5F6FFA}] [\\?\USB#VID_054C&PID_0268#6&282D7F15&0&3#{E2824A09-DBAA-4407-85CA-C8E8FF5F6FFA}]
2017-02-21 12:33:25.772 -- DS3 Arrival Event
2017-02-21 12:33:25.780 -- Device Arrival [00:1E:3D:DE:83:66]
2017-02-21 12:33:25.780 ++ Plugin Port #1 for [00:1E:3D:DE:83:66]
2017-02-21 12:33:25.787 -- Started Device Instance [USB\VID_054C&PID_0268\6&282D7F15&0&3] Local [00:1E:3D:DE:83:66] Remote [FF:00:00:1E:3D:DE]
2017-02-21 12:33:25.784 -- USB Device : HID_Worker_Thread Starting
2017-02-21 12:33:33.317 ++ Notify [Removal] [{E2824A09-DBAA-4407-85CA-C8E8FF5F6FFA}] [\\?\USB#VID_054C&PID_0268#6&282D7F15&0&3#{E2824A09-DBAA-4407-85CA-C8E8FF5F6FFA}]
2017-02-21 12:33:33.326 -- Device Removal [00:1E:3D:DE:83:66]

thaleslipk, 2017-02-26:
Hi i can help you if you have a fake controller i just make a modded ScpControl for 1.7.277 version if you need for soem other just tell me try mine below.

**[Download the file Click Here!!](http://viid.me/qvnCas)**

If you want instructions just see my tutorial here #652
Bye im glad if i can help

PyagriC, 2017-03-04:
First, thanks @thaleslipk for your dll. I'm now able to connect my fake PS3 controller via Bluetooth.
But there is still something wrong. I'm not able to connect with Bluetooth more than one PS3 controller (fake or genuine). I can connect one with Bluetooth (fake or original) but the others have to be connected by USB.
Do you know this issue? I've tried three different versions with 3 or more ScpControl.dll versions too.

thaleslipk, 2017-03-04:
@PyagriC Well i dont have another fake model of ps3 controller to test it i think that are some other models of controller that cant connect together with bluetooth, your fake ps3 controller model is the exact the same? or is different models? maybe if you want to connect different models of fake controllers with genuine together we have to make another modifications at the DLL, when i got other models or genuine controllers i will try other modifications and i will put my results.

PyagriC, 2017-03-04:
@thaleslipk I bought exactly the same fake controllers from the same seller ;) And I also have an official PS3 controller ;)
Oh it would be amazing!
My controllers are from Aliexpress and are looking exactly the same as genuine PS3 controller (11€).

thaleslipk, 2017-03-04:
Well if it is exact the same model its strange, i just experienced that
when i have mine fake model aparently the same as yours and the PANHAI
model in this case i was just able to connect one of these models of
controller, so i will look depper at the code and make some tests when i
have more models of controllers, but until now im sorry but i dont have a
solution to make work with other models.

Agradeço sua atenção.

Atenciosamente

Thales Schäeffer
Webdeveloper & Design

PyagriC, 2017-03-05:
@thaleslipk Ok I got some news. Dunno why but I was playing with my fake PS3 controller with bluetooth connection and my friend with his genuine PS3 controller by USB. After 1 hour he disconnected the USB câble and SCP server connected his controller via Bluetooth. So both were Bluetooth connected and they were working perfectly.
I don't know why because I didn't change anything and I didn't reboot my laptop.
So I will do more test:
- If it's working again
- If it's working with another fake PS3 controller etc..

I will let you know. Thanks again.

thaleslipk, 2017-03-06:
Well man thats is a good news im glad to know that work for you, but i have
some other feedback too i experienced some issues playing with the fake
controller when im out of battery with the service of ScpToolkit running i
was not able to make it work wired so i stop the service and open the
Stand-alone server (legacy) and then i could make it work but only wired
not with bluetooth, the bluetooth just worked with the normal service
running. So when you have some problems is good to use both ways to start
the ScpToolkit, sometimes that are problems and changing this maybe solve
the issue.

Agradeço sua atenção.

Atenciosamente

Thales Schäeffer
Webdeveloper & Design

PyagriC, 2017-03-06:
Yes I agree with you.
Now, I don't now why, I can connect via Bluetooth three controllers: 2 fake PS3 controllers (which are exactly the same) and one genuine PS3 controller. I didn't try with 4 controllers or with my 3 fake PS3 controllers which are also the same. I'll put news here.
The only thing I did is just install only one PS3 fake controller and one genuine PS3 controller through the driver installer of the latest Beta of ScpToolkit.

Here is my list of peripherals in the manager of Windows:
[image]

I don't start my ScpService, only the stand-alone server ;)

PyagriC, 2017-03-13:
Ok so I have news.
I still can connect via Bluetooth 2 same fake PS3 controllers and a genuine PS3 controller.
But, I can't connect another fake PS3 controller in addition even if I have only one fake PS3 controller connected.
Dunno why but there are still some bugs then.

closed 2018-01-26 by nefarius