nefarius/ScpToolkit #642: Scpcontrol.dll modification

Dienfive, 2017-02-22:
Dear All,

I have this log:

2017-02-21 12:33:25.772 ++ Notify [Arrival] [{E2824A09-DBAA-4407-85CA-C8E8FF5F6FFA}] [\?\USB#VID_054C&PID_0268#6&282D7F15&0&3#{E2824A09-DBAA-4407-85CA-C8E8FF5F6FFA}]
2017-02-21 12:33:25.772 -- DS3 Arrival Event
2017-02-21 12:33:25.780 -- Device Arrival [00:1E:3D:DE:83:66]
2017-02-21 12:33:25.780 ++ Plugin Port #1 for [00:1E:3D:DE:83:66]
2017-02-21 12:33:25.787 -- Started Device Instance [USB\VID_054C&PID_0268\6&282D7F15&0&3] Local [00:1E:3D:DE:83:66] Remote [FF:00:00:1E:3D:DE]
2017-02-21 12:33:25.784 -- USB Device : HID_Worker_Thread Starting
2017-02-21 12:33:33.317 ++ Notify [Removal] [{E2824A09-DBAA-4407-85CA-C8E8FF5F6FFA}] [\?\USB#VID_054C&PID_0268#6&282D7F15&0&3#{E2824A09-DBAA-4407-85CA-C8E8FF5F6FFA}]
2017-02-21 12:33:33.326 -- Device Removal [00:1E:3D:DE:83:66]

After reading lots of comments, i found out that i need to modify scpcontrol.dll file in order to make it works thru bluetooth. However i dont know how to modify it, do i need to download source code, and modify them and create new .dll file? If so, which source code i need to modify? please tell me.

Thank you very much!!!

thaleslipk, 2017-02-26:
Hi i can help you if you have a fake controller i just make a modded ScpControl for 1.7.277 version if you need for soem other just tell me try mine below.

[Download the file Click Here!!](http://viid.me/qvnCas)

If you want instructions just see my tutorial here #652 
Bye im glad if i can help

Dienfive, 2017-02-28:
Hi @thaleslipk , thank you very much. but i follow installing ScpToolkit v1.7.277.16103-BETA. but the all controller only work thru PC, not bluetooth. please help to check

thaleslipk, 2017-02-28:
@Dienfive Hi, but  what controller do you have? its a fake PS3, 3rd partie? or genuine? because i make it work on bluetooth a lot of models i know there are a lot of different models out there but i search your hid and found a clone model of PS3 controller...but you used the modded dll i left on my post #652 ? Maybe your model have a similar connection of the PANHAI ones so try it this other **[modded dll here in this link](http://adf.ly/1kX3RZ)** maybe works with you.
**Note:** Another thing, is good to mention as your model controller on another emulator are not working because the PS3 console are being turn on and paired first than the Scp on the PC so if you have a PS3 console you have to turn it off unplugging from the power source.

Dienfive, 2017-03-02:
@thaleslipk Hi, i made this one connected thru bluetooth but for rumble testing only, other function not work. Can  you please check?

[image]

closed 2018-01-26 by nefarius