nefarius/ScpToolkit #643: Driver installer stuck

mtfck, 2017-02-22:
My issue seems to be a lot like Keefaetana's [issue](https://github.com/nefarius/ScpToolkit/issues/177). The driver installer simply gets stuck on the "Starting driver installation, this might take up to five minutes..." step. Reinstalling and rebooting did not help. Any way to solve this?

I'm also running Windows 10 x64.

Thanks

mtfck, 2017-02-22:
Nvm I just needed to install without driver signature verification and it worked perfectly thank you.

closed 2017-02-22 by mtfck

dill0wn, 2017-09-27:
Gah! I'm having the same issue, it persists even if I disable driver signature verification :(

shahzaibjamal, 2017-11-30:
@dill0wn any luck with this issue. I also tried disabling driver sig verification but it is stuck there