nefarius/ScpToolkit #644: Controller (remote) with a mac address of FF:FF:FF:FF:FF:FF

Lyoud, 2017-02-23:
As the title, what is the meaning of that? My host mac address is as usual. Wired ds3 works, yet bluetooth is  not working.

thaleslipk, 2017-02-26:
Hi i can help you if you have a fake controller i just make a modded ScpControl for 1.7.277 version if you need for soem other just tell me try mine below.

[Download the file Click Here!!](http://viid.me/qvnCas)

If you want instructions just see my tutorial here #652 
Bye im glad if i can help

Lyoud, 2017-03-02:
Hi there, I now only see your comment. After tried all the way you shown, even deep searched GUIDs. Yet the server still with 00000000 none bth service when conected with bluetooth.

closed 2018-01-26 by nefarius