nefarius/ScpToolkit #648: My bluetooth adapter is not supported, help?

curtiola, 2017-02-25:
any chance you can make this work with my bluetooth adapter?
I'd really appreciate it :)

it's a micro star
USB/VID_0DB0&PID_A871\7&5CC54C7&0&1

curtiola, 2017-02-25:
I solved the problem I was having, now both DS3 and DS4 connect via bluetooth with SCP Toolkit
I downloaded the SCP-DS-Driver_package-1.2.0.160 from this website:
http://forums.pcsx2.net/Thread-XInput-Wrapper-for-DS3-and-Play-com-USB-Dual-DS2-Controller

once downloaded I went into the ../bin/Zadig directory
opened the Zadig program, click device>load preset, load the "bluetooth" preset
then click options>list all devices, selected my bluetooth adapter from the list and replaced the driver
afterwards I loaded the DS4 preset, went to options>list all devices, chose "wireless controller" and replaced the driver and repeated the process for the DS3 controller

I hope this helps someone else who had my problem :D

thaleslipk, 2017-02-26:
Hi i can help you if you have a fake controller i just make a modded ScpControl for 1.7.277 so much better than the old version 1.2 if you need for some other just let me know but try mine below first.
Works like a charm for 3rd partie controllers.

**[Download the file Click Here!!](http://viid.me/qvnCas)**

If you want instructions just see my tutorial here #652
Bye, im glad if i can help.

cejdev, 2017-03-01:
Brilliant curtiola - bluetooth works great now! 

Using the current stable installer, Bluetooth wasn't working after restarting, so I cleaned it all off and upgraded. The beta driver installer wasn't recognizing my Intel Wireless/Bluetooth card (not faulty, and factory drivers installed). It worked before on the stable version so no idea why it's suddenly "not supported". I finished out the driver installation, and it all worked minus Bluetooth. I then followed curtiola's instructions (skipped the DualShock3/4 preset install part), and it works great!

rougetimelord, 2017-03-02:
Hey, since you solved your problem could you close the issue? Just to declutter 😄

cejdev, 2017-03-06:
I spoke too soon. Worked great for a few days. Suddenly no longer detects it. Sounds about right. About to sell this thing and just get an Xbox controller, ridiculous.

thaleslipk, 2017-03-06:
Hey man try both Scp normal service if is not working these way stop the
service and open the Stand-alone Server (legacy) for me solve some problems
when is working just wired or just with bluetooth.

Agradeço sua atenção.

Atenciosamente

Thales Schäeffer
Webdeveloper & Design

closed 2017-03-30 by curtiola

michaelstchan, 2017-06-23:
@curtiola did you install the drivers with toolkit before or after using zadig?

curtiola, 2017-08-29:
@michaelstchan I installed the toolkit drivers first, then replaced them with zadig