nefarius/ScpToolkit #650: PS3 disconnecting while playing

Cooluser10, 2017-02-25:
Hello!

I have a super weird issue. I'll be playing Dark Souls for a bit and then the driver will tell me the controller is unplugged, then it takes awhile before it detects it again. The weirder thing is, if I play For Honor, the controller will disconnect on every input. Leaving it idle now, it hasn't disconnected itself at all. All this time I can confirm that my USBs are securely connected. 

Please help me, thank you very much.

GrimOakheart, 2017-03-11:
I have the exact same problem. Can you reply to this thread if you find a solution? 

Ty in advance.

Aya094, 2017-03-19:
By chance, when it disconnects does the control rumble? If so, try turning it off

closed 2018-01-26 by nefarius