nefarius/ScpToolkit #652: Fake PS3 Controllers Bluetooth 00000000 problem SOLUTION!

thaleslipk, 2017-02-26:
Hi im bringing not an issue but a SOLUTION so as i know every almost user came here first to see a solution for their problems im writing that.
First of all thanks to @nefarius   for this great software and his great hard work.

The file ScpControl.dll here is for the version 1.7.277.16103 so first check if your version is that and this modded file can fix a lot of fake PS3 controllers that one that are being selled as an original one but its not.
**WARNING:** THIS FIX MAYBE NOT WORK FOR EVERY FAKE CONTROLLER AND MAYBE CAN BRAKE THE CONNECTIVITY WITH THE GENUINE ONES, YOU HAVE TO TRY AND TEST.
BEFORE YOU DID ANY MODIFICATION PLEASE MAKE A BACKUP OF THE FILES YOU HAVE TO CHANGE.

**1-** Download the file attached is on .zip format extract the **ScpControl.dll** copy and paste on the ScpToolkit folder instalation and replace it, start the **ScpToolkit Stand-alone Server** and connect your controller with the USB press the PS button when the LED stop flashing wait a little then disconnect the USB cable and  its DONE!! Bluetooth connected, cry a little bit and thank me for that :p.
You have to do that with SCP service stopped and none of the Scp applications running ALL STOPPED.

**Note:** If you replace the file ScpControl.dll and you cannot make it work follow these next steps.

**Optional Steps**
2- Go to your Device Manager and search for your **bluetooth adapter** right click on it open the **Properties** look at the **Hardware** tab and click **Properties** again and **Details** tab click on the dropdown list search for **GUID class device** copy the whole string and paste it on the **ScpControl.ini** file on the folder instalation at the **[Bluetooth Dongles]** in this format 
i.e (DeviceGuid = {The Number of you GUID here}).
3- Repeat the 2 step until it came to properties of your bluetooth adapter and on the **Details** tab on the dropdown list search for **Hardware IDs** Copy the whole string of the two and paste it on the ScpControl.ini file at the **HIDs of supported bluetooth dongles** in this format
i.e (HardwareId = USB\VID_0BDA&PID_8723&REV_0200
     HardwareId = USB\VID_0BDA&PID_8723) *Thats an example use it your device ID sure right!!
4- Repeat the 2 and 3 step for your **PS3 controller** in the device manager selecting it intead of your bluetooth adapter and paste the strings of the **GUID** at the **[DualShock 3 Controllers]** and the **Hardware IDs** to the **HIDs of supported DS3 controllers**  in the same format as described before on the ScpControl.ini file.

So thats all, i expect you all can fix the problems wth bluetooth of your controllers with this modded fix just to remember:
VERSION OF THE FILE IS  **1.7.277.16103** **SO JUST WORK AT THE SAME SCP TOOLKIT VERSION!!!!!!**
Bye ;)

EionRobb, 2017-02-28:
Can you provide the source of your modifications please

EionRobb, 2017-02-28:
[ScpControl - Fix Fake PS3 Controller By Thales.zip](https://github.com/nefarius/ScpToolkit/files/806126/ScpControl.-.Fix.Fake.PS3.Controller.By.Thales.zip)

thaleslipk, 2017-02-28:
@EionRobb I edit the workaround code for fake controllers on BthDs3.cs
`{
                    _hidOutputReport[0] = 0xA2;
                    _hidOutputReport[3] = 0xFF;
                    _hidOutputReport[5] = 0x00;
                }`

Thats it.

Erquint, 2017-03-03:
@thaleslipk Man, I haven't tried your fix but I just wanna say the way you distribute it is a bit of a problem.
The proper way would be to fork this git, since keeping an issue open for a non-issue is kinda backwards and it's not very discoverable.
People online are silly scaredy cats and are in fact overreacting but you give them a reason by employing these guerilla tactics of sharing executables in a discussion. Also, there's nothing technically wrong with using AdFly as a link shortener for any sane person but, again, normies ≠ sane and see it as a top-level security threat.
Anyway I get that you don't wanna fork for one line of code modification but it hurts seeing you trying to help and getting that kind of treatment:
http://i.imgur.com/Fgi9HFR.png

thaleslipk, 2017-03-03:
@Erquint Man i really dont matter people downvoting my posts, only problem is when you lost a chance to fix your problem cause you was scaried and don even tried, cause i modificate a little bit of code was actually 3 lines but i build the DLL for 1.7.277 and a lot od people are downgrading the software to older and not so good versions to work with their fake ps3 controllers i just tried to help to mantain people on the latest and best release and have the workaround working for them.
Another thing is im not sharing an executable its a DLL people can open it on dnSpy for example and inspect the code no problem, above i provide the modifications if dont like it or are afraid to download the DLL i shared, make your own modifications and build your own DLL am not forcing anyone here.
I build that DLL for people who cant make a DLL, people who doesnt know how to do that, i tried to help these people, other people who have minimum programming skills, make your own!! :)

thaleslipk, 2017-03-03:
Again i open that as an issue because i know its the first place where people search to fix their problems, the best way to help and let be seen.

Erquint, 2017-03-03:
Dynamically linked libraries are very much executable by definition despite what the extension says — those are arbitrary.

mmbossoni, 2017-03-06:
Worked like a charm.
I have one of these plagued fakes.

thaleslipk, 2017-03-06:
@mmbossoni Really good to know, if you make ir work with different models, brands of controllers or genuine ones just put your feedback here to get to know what is really working, i get some feedback on other threads that some genuine ones are working with the fake controllers together with bluetooth :)

mmbossoni, 2017-03-06:
Tried together with a genuine controller and it worked fine.

thaleslipk, 2017-03-06:
@mmbossoni Thanks for your feedback is really good to know!! now i can buy it without being afraid hehe :D

thaleslipk, 2017-03-06:
Dynamically linked libraries are very much executable by definition despite the arbitrary extension.
In the sense that they can be executed and are made to be executed. Where executing is mapping a process image to memory for the processor to execute. They are only different from PE(exe)s in their entry points.
Technically you could insert any malicious code into a DLL that would get loaded by SCP and get it to execute on people's machines but I don't think you are here to infect anyone.

its the first place where people search to fix their problems
I somehow doubt this. Most are probably using Google and your issue isn't high up in the results for me.

P.S.
Do you think your fix will work better than the original Scarlet Crush Productions solution did for my knock-off? (#628 (comment)) Key point I want to know is if Nefarious' tool with your fix is gonna work better for me than the original tool from PCSX2 forums in terms of compatibility and performance.

@Erquint Well, i said before that DLL are not executables because it cannot be executed directly right, so its not by definition an executable, but i know it can execute instructions thats really obvious to make a discussion about it right hahaha, because that is actually what the DLL is doing at the ScpToolkit :p.

The search for problems part, what can i say is... yeah you right! first thing that people do is googled things i know that, but when i googled issues with ScpToolkit i end up here at the GitHub so its easier just search for solutions here at the **Issues** tab of the project sounds obvious huh? ;), another thing is if you have the ScpToolkit you probably download it here, so the section named **Issues** should be your first place to go, dont you agree with me? or you want me to have a blog with a post about it so much well ranked to be in the first page of Google? Well i dont know if you know about it, but rank to the first page at the Google its not that easy to do so much faster, so the better way i found to help was opening a issue with that title, so if someone **googled scptoolkit bluetooth issues** probably like what happened with me, will gonna find another open issue here at the GitHub and if this person look all the issues maybe will find mine or my reply on the other issues related to that one that i wanna help to fix,ok?

About the softwares, if the older DLL works for you with your knock off controller, well ok good for you, but my experience was not good i always have to reinstall the drivers after a system restart, so for me was a headache and the only software that i can recommend because its easy to install, well ordered, well explained and as i could see of the code looks a really clear code and well made i did not monitor the performance compared to original Scarlet Crush but i used both and the one that worked for me it was the Nefarius ScpToolkit. I think if you talk about consume of memory or CPU, the Nefarius should be heavier because it install more elements, you know its a cleaner modern GUI, run on background, have a pop up notifications so it was an evolution of the Original Scarlet Crush software and actually for what it provides its extremely light and soft and working with my fix do not change anything i did not experienced problems in performance on the system and on the controller as well, i dont have a delay in buttons its just like play with bluetooth on PS3. But if you got your controller working well you do not need to change to mine, i provide this DLL for people who cannot play with Nefarius ScpToolkit v1.7 so they do not need to downgrade to older versions.

nendraharyo, 2017-03-21:
Hey thanks dude it works! but there's a problem. I need to open the SCPtoolkit Settings Manager but it gives me the error message as you can see in the screenshot. Please help :(
[image]

hyptechdev2015, 2017-03-28:
great works, thank you @nefarius so much.

thaleslipk, 2017-03-29:
@nendraharyo Man, i dont have time now to inspect this problem on the code, so the solution i have so far for you is just replace this modded dll by the original dll and open the settings change what you want and after you finish stop the service replace it again by the modded dll for fake controllers and its done.

Nudg, 2017-04-03:
hello, is this thread for OSX or Windows?

adhany, 2017-04-06:
I have installed latest beta scptoolkit everything was installed . But when opened scp server there show host adress error ? Try to clean driver using wipe tool and reinstall but no change . Please help

thaleslipk, 2017-04-07:
is for Windows

Agradeço sua atenção.

Atenciosamente

Thales Schäeffer
Webdeveloper & Design

krisrgarrido, 2017-06-12:
Hello everyone, I got the bluetooth working before with my fake ps3 controller, but when my pc bluescreend and restarted I got the "Fake Dual Shock 3 found, trying workarounds." I tried replacing the DLL again and also tried reinstalling, still no luck. I also tried running the SCP driver installer but now it just crashes.. any help?

dwikiprayoga, 2017-07-07:
@thaleslipk Thank you very much! This is working flawlessly. I should've discovered this sooner. Now I can play without worrying about the cable.

daleylife411, 2017-07-11:
I believe I might have a fake controller.
But, whewn I try to place the file that is here, I get "you need permission to perform this action." It will not let me replace the original file.

thaleslipk, 2017-07-11:
Man, for replace the file you should stop SCP toolkit application that is
running and the service of SCP. When everything is stopped you replace it
with no problemas surge you have to use an admin account privilegies.

daleylife411, 2017-07-11:
Okay, well, I just tried it--- I closed the SCPtray and I see no indication it's running, and I still can't change the file. I am the admin of the computer.

mmbossoni, 2017-07-11:
You must stop the service
On Task Manager, go to services tab and stop DS3 something service
Then you try to replace it.
Tray is just, well, a tray showing server messages.

daleylife411, 2017-07-12:
Okay, I found the service in Task Manager. But, when I try to stop the service, I get "access denied."

daleylife411, 2017-07-12:
UPDATE:

What I ended up doing was just deleting the original file and then placing the edited file (from here) into the folder. I  installed and now, IT WORKS. I have not tried bluetooth, but, I'm just happy I can play my games again.

Still have no idea why the folder wouldn't let me replace files or why the DS3 service couldn't be stopped. Again, I am the admin, so, that was weird.

h3mPi, 2017-08-04:
@thaleslipk  OMG I love you! It works!
My fake dualshock 3 controllers (called Gasia GamePad) stopped working on my laptp after i reinstalled scp toolkit(1.6) and only my original ds3 continued to work..  On my Home PC they continued to work and I just couldn't wrap my head around what happened.. Months of reinstalling and trying different versions followed, I have before tried your .dll but I guess I didn't notice that it is for 1.7.. 
So today I tried the .dll again (only did first steps) and now my shitty Gasia Gamepads work again! 
Thanks so much!!

thaleslipk, 2017-08-04:
Well man, so good to know im accepting gifts if you want lol 😆 kidding,
good to know another model that i do not had in the base works, thanks for
your feedback if you do another tests with other models or with genuine all
together with fakes please send the feedback of your experience to help
another users and the community, enjoy!😉

h3mPi, 2017-08-04:
I would definitely send you 5eur or something for a beer/coffee if you give me your PayPal address/link :)

I have 2 gasia fake ds3, 1 original ds3 and 1 Xbox controller  connected right now via bluetooth dongle/xbox wireless receiver and they all get recognized by Dolphin Emulator. So we will play this evening.. :)
If you need anything specific like logs or something, just tell me!

thaleslipk, 2017-08-06:
Well a coffe will be good hehe thanks if you want my PayPal is:
thalesschaeffer@gmail.com  but no need, just if you want 😉.
Good to know everything works together with you, a time ago i receive a
feedback that 2 genuine DS3 with 2 fakes, 1 of the fakes are not working
together with all of that just separate or usb i couldn't test it out if
you can do that set up will be good to know if works, well thanks man and
keep enjoying gaming im glad to know i help.

xanscorp, 2017-08-21:
Kind of humorous that you thank them for their hard work...yet you're the one who had to find a fix for it because they didn't.  Additionally the "fake" issue may be a glitch in the software itself as my  PS3 controller direct from Sony is shown as fake.  Basically a genuine controller from the company that makes it is fake...according to the software.  I'm trying to resolve an issue where the connection turns off...then refuses to reconnect unless I uninstall anything associated with SCP and reinstall after rebooting.  Sounds annoying and incredibly frustrating?  It is.  Wish they'd fix this crap, but yeah...that's not likely to happen.  Others will always have to clean up the designers' mess.

thaleslipk, 2017-08-21:
Well i thank cause he want to send some donation and for the feedbacks to
grow the base of known devices.
What version of SCP Toolkit are you running? Be sure is the one mentioned
above, i just try this on Windows 7 but in others should work, if you have
a genuine controller should work just fine if are shown as Fake in the
application maybe is not actually genuine a lot of places are selling the
console with perfect copies of the genuine controller because its not
manufactared by Sony anymore in a long time ago.

xanscorp, 2017-08-21:
Have to check when I log back on my system.  I had  to reboot after uninstalling...also had to walk away before the annoyance cause me to damage stuff.  (Inanimate objects cannot press aggravated assault charges so were my system cheap or if I still had anything easily replaceable, it'd have gone flying.)

xanscorp, 2017-08-21:
I'd wager on the software being the issue.  As I mentioned, my controller is direct from Sony, not a reseller.

thaleslipk, 2017-08-21:
Just another thing if you have your controller working on the real console
you have to turn the power off unpluged from the energy, cause even when
you turn off your console PS3 or PS4 that can put inteference between the
bluetooth with your PC and controller.

xanscorp, 2017-08-21:
I know.  That's one of the first things I did.  My issue is the controller timing out and will not reconnect.  Of course since uninstalling and reinstalling require a reboot...doing anything prior to save points that would cause me to be away for more than 2 minutes is next to impossible.

xanscorp, 2017-08-21:
I'm tethered using the USB cable...all other USB devices do not time out and cease to function, though.

h3mPi, 2017-08-22:
Well first of all, that's not a nice way to start a conversation.. :)
This software is provided free and open source, so if you have problems with it you can look at the code and try to fix it yourself or you better ask nicely and patiently if anyone can help.
Thaleslipk isn't even the maintainer of this add-on and still he made and uploaded his DLL to support more fake controllers in his free time to help others..

Second, are you 100% sure your controller is legit? Bought mine too in an electronics store and they were supposed to be original, but when I opened them up and looked on the inside (battery was really tiny) they clearly weren't.. (Had a hard time but eventually got my money back)

What do the logs say? Does it work on other systems? If you have a Linux box try it with sixad/qtsixa

xanscorp, 2017-08-22:
Actually it is a neutral way to start a conversation.  If I gave your a bunch of parts that were just scrap and you make it something workable, if you thanked me, I'd wonder way.  I only handed you scraps.  And  two.  Read correctly.  Mine came from Sony...making your statement of "Bought mine too on an electronics store" proof that you assumed instead of reading.
 
Yes, my  controller functions if run through VirtualBox on Linux, but as I mentioned, I do not use it as a Bluetooth, only directly wired via USB.

GOESTYN, 2017-09-29:
it worked for me! i just registered here to thank you so much, i have been trying for weeks and nothing, but you. you bring me the fix ty ty <3 <3

Goonersheart, 2017-10-02:
Didnt work for me, help please :(

MountyBounty, 2017-10-07:
My controller, according to device manager is a PS(R) Ga'epad and my usb bluetooth dongle is a 'Bluetooth V2.0 Dongle'. The dongle works with genuine controllers so I'm guessing it's not the dongle. I have added the two GUID's of both the PS(R) Ga'epad and the 'XBOX Controller' that show up in my device manager. And also of course the guid of the bluetooth dongle as well as the hardware id's of both the listed controllers and bluetooth dongle to the ScpControl.ini but no luck. I can use it with cable but trying it via bluetooth I still get a 'Fake ds3 controller found, trying workaround' notification with no luck. Any suggestions?

thaleslipk, 2017-10-08:
Maybe your controller is a PANHAI one so check the post above and look at
the obaervation i did about that there is a link to download the DLL
modified to PANHAI models, remember to aways check the version of the file
abd the versuon of your SCP Toolkit to match as the same version.

MountyBounty, 2017-10-09:
I love you man. I actually tried that file but while also having changed the scpcontrol.ini and then it didn't work but after reinstalling and not touching the scpcontrol.ini and just replacing the scpcontrol.dll with the file you provided it worked instantly. Thank you so much :D

thaleslipk, 2017-10-09:
Good to know, is importante to say you just edit the scpcontrol.ini when do
not works normally in the first attempt because maybe you messed up the
code without notice.

alansublime, 2017-10-20:
friends the solution for gawron89 https://github.com/nefarius/ScpToolkit/issues/153

whise, 2017-10-24:
here is the latest scptoolkit with fake controller and gyro support. 

I compilled it myself using this patch: https://github.com/rajkosto/ScpToolkit/releases

Get it here: https://mega.nz/#!EFN32J6R!JlJ1-b2mRARJFxa6nMq_tX3MQUzXv2tSHhGbEpbw0MQ

mikegonzalez2k, 2017-11-03:
I just chatted with Nefarious online.  He is actually in the process of releasing a new application to fix many of these kinds of issues.  The DS3 random button press issue is a problem with the driver initialization every so many seconds.  It cycles and randomly triggers.  He's well aware of the issue as it's been plaguing SCP since version 1.6 back in 2015.  Anyhow he fixed it in a new release.  But that isn't out yet.  Keep an eye on his github, hopefully it will hit soon.

cantdobluetooth, 2017-11-13:
bro i tried both steps and still didnt work, it still say bt 0000000 none, i literally make an account just for this, so stressed out i because i tried hours and hours searching different forum for this problem but still didnt work, is there any other options ? or the dev is working on it ?

thaleslipk, 2017-11-13:
Hi, well the steps you should take s the one above, maybe do a clean wipe
uninstallation, remove the drivers os SCP bus and for bluetooth scratch
from zero and start again doping the steps or maybe you are using a version
of the program different of the DLL modified here check above the note
about the version, or you have a PANHAI modelo of controller so you have to
use another DLL i mentioned that above in the note, so read everything and
try it out.

alansublime, 2017-11-14:
@cantdobluetooth Friend... download: [http://www.mediafire.com/file/0wc4zhbuihyc64c/Control+SP3+x64.rar](url)
1. delete all previous installations
2. Install dotNetFx40_Full_setup
3. Install dxwebsetup
4. Install vcredist_x64
5. Install Xbox360_64Esp
6. finds the path SP3 Control x64 \ ScpServer \ bin
7. Right click on: ScpDriver.exe ... properties ... general (UNLOCK), apply and save
7.1 Connect to control
8. Execute as an administrator
9. Check all... Install All.
## VERY IMPORTANT:
### a. Open Control SP3 x64\ScpServer\bin\Zadig... execute as an administrator "zadig.exe" (do not forget to unblock .exe in properties) 
### b. Click in Device... create"New Device"... Click in Device..."Load Preset Device"... open "Bluetooth.cfg" (Control SP3 x64\ScpServer\bin\Zadig)
### c. Click in "List all Devices"... Select your bluetooth device... Click Install Driver. Close.
10. Open ScpServer.exe as an administrator (do not forget to unblock .exe in properties)
12. Click on "pair" or run..,if not available this option: open  "ScpPair.exe .exe" (do not forget to unblock .exe in properties) and copy the mac address of the Bluetooth device and pair
13. Disconnect Control...
14. wait a moment, if you continue 000000 press again the "P" button of the control
15. If you still continue 0000000, close the "ScpServer.exe" application
16. It should work if you do not let me know, to try more things.

:D

cantdobluetooth, 2017-11-18:
@alansublime the pad doesnt show on scp server

cantdobluetooth, 2017-11-18:
@alansublime NVM MAN ITS WORK !! WTF U OWE MY LIFE MAN

alansublime, 2017-11-18:
@cantdobluetooth 
Did it work?

gojpgo, 2017-11-19:
@thaleslipk I tried using 1.7.277.16103 but I couldn't install bluetooth drivers through it. So I reverted back to 1.6.238.16010. USB works fine and all. Bluetooth connects as Bluetooth 000000000 none. It'd be nice if you can modify ScpControl.dll for 1.6 version. I tried to modify it but I couldn't. Any help would be great!

gojpgo, 2017-11-19:
Here is the ScpServer log, if it helps
[image]

thaleslipk, 2017-11-19:
i don't have the development tools this time to make it again but i think i
have one of the 1.6.xxx dll modded but i don't remember wich version
exactly is, maybe will not work if is not exactly the same i will look it
up.

gojpgo, 2017-11-19:
@thaleslipk thank you for your input. SCP toolkit 1.6.133.15324 is working perfectly well with bluetooth!! Thanks again..

gojpgo, 2017-11-19:
Although It'd nice to have rumble over bluetooth.. It'd be great if you can fix it. If you have time ofc!

alansublime, 2017-12-10:
Help friends no connect,"2 control ps3 Fake", error. 

[image]

ERROR: Could´t plug virtual device 2: System. ComponentModell.win32Exeption (0x80004005): The operation completed successfully
INFO: Plugged in port #2 for 00:26:5C:92:9E:05 on virtual Bus 
L2CAP_Disconnection_Request Handle [System.Byte[]]
L2CAP_Disconnection_Response
HCI_Disconnection_Complete_EV [05]

Help Friends, Please

closed 2018-01-26 by nefarius