nefarius/ScpToolkit #653: After installed just doesn't show up or anything as a controller

damron17, 2017-02-26:
I got no idea what I did but I'm using the latest version. The only thing I did before is I think accidentally delete the 360 driver that came with windows 10 but I went and got the windows 7 one and all that but everytime I go to install everything it goes like it works and even makes the sound and everything. Th stand alone server even sees the ps3 controller. It just won't switch it over to emulate the 360 one anymore for me. The older version made by Scarlet Crush still works for me but I've uninstalled that one and used the wipe utility before reinstalls everytime.
[ScpToolkit_Log-Package_damro_1488148890.zip](https://github.com/nefarius/ScpToolkit/files/802404/ScpToolkit_Log-Package_damro_1488148890.zip)

thaleslipk, 2017-02-27:
Man you have to Clean Wipe Utility, then you go to your Device Manager and search for some drivers wich possibly was left behind, search on **System Device** see if theres some **Scp Virtul Bus Driver**  and on **Universal Serial Bus Devices** any driver **with the Scp name on it, bluetooth or controller**  so manually uninstall the driver, its good if you can uninstall these xbox 360 driver too and install it again then restart computer and install the version 1.7.277.16103-BETA follow the steps to install see if your controller shows on the instalation and your bluetooth driver too just follow normally if you can get this work maybe you have a 3rd partie controller i dont know try to see whats your controllers i have some modded dll wich can work.

damron17, 2017-02-27:
I don't have a 3rd party controller. I'm using a ps3 controller. I've done the device manager stuff uninstallation stuff via what it shows on wiki here shows. Anything I saw that was there I saw associated to my controller other then the xbox driver. I thought that needed to stay there.

thaleslipk, 2017-02-27:
well if you sure your controller is genuine not other branded or fake ps3, you problem is the driver for xbox as you mention you are using a windows 7 and you got a windows 10 so see if this video is similar to your case maybe you still have the windows 10 driver for xbox but its not installed so this video can help you to search for the latest driver.
https://www.youtube.com/watch?v=0V6b75ZDcCI

damron17, 2017-02-27:
So that didn't work either. I uninstalled the stuff. Reverted the xbox driver to the one in the video. Reinstalled the controller and the scptoolkit after a restart. Installed the driver and nothing showed up in the controller thing.

Jahron1, 2017-02-28:
I'm having a similar problem trying to install on a very old laptop running Windows XP.  Before I install using SCPDriver, the laptop shows that it sees my PS3 controller.  When I install the drivers without forcing install, nothing happens, SCPServer doesn't read my controller or anything, but when I check the "Force Install" box, SCPServer can read my controller and inputs, but my laptop can no longer see the controller at all.  I am unsure of what to do.  The weird thing is, when I plugged in the controller and did the install for the very first time, it was 100% working, I could move my character in the game I was trying to play and everything.  After I unplugged it and plugged it back in, it just stopped working and hasn't worked since

Jahron1, 2017-02-28:
I'm not sure I understand how to use this Clean Wipe utility.  I've downloaded the toolkit folder, how do I do the clean wipe?

Jahron1, 2017-02-28:
I honestly don't even know what I did that made it work but it works now.  I went to device manager and uninstalled "USB Human Interface Device" (which was for my PS3 controller).  THEN I went and tried to install the drivers from ScpDriver, had to force install them.  Now it recognizes my controller as an Xbox 360 controller, reads inputs, and works with my game.  Hopefully it stays this way

thaleslipk, 2017-02-28:
@damron17 Hi, maybe if you do like @Jahron1 did uninstall all the HID drivers not just revert the driver for xbox but all of USB HID that is shown on the device manager just be carefull with other USB devices that you use and then restart PC and plug your Controller on the USB and wait for the auto instalation driver if everything installs correctly try Scp again, good luck.

damron17, 2017-02-28:
I've did that and it doesn't work either. I looked through all my HIDs for anything that looked like it was associated to a controller even in hidden devices and uninstalled that device. I unplugged my controller did a restart and when I plugged my controller back in after the restart and tried to install the scptoolkit again it just says its a ps3 controller still and doesn't change it to an xbox 360 controller.

closed 2018-01-26 by nefarius