nefarius/ScpToolkit #656: My USB mouse and keyboard don't work anymore

SterreStarGames, 2017-03-02:
When i installed everything my mouse and keyboard froze HELP!

Fred1rst, 2017-03-28:
Well then well then, almost started a topic but it seems I'm not the only one to have this issues!

I read here that many people have overridden their mouse and keyboards drivers.
How the heck am I supposed to ":Revert the changes by hand in Windows Device Manager and revert to the correct HID drive" (nefarius 2016-jan-09)

Please help my mouse is dected as keyboard HID and vice versa...

Fred1rst, 2017-03-28:
UPDATE: That's stupid, after a week of trying, 
The mouse and keyboard need to be plugged in in order to delete the drivers in "libusb" from the device manager... If not plugged the libusb "gaming mouse" just disapears.
Every step is useless unless the HIDs are plugged in.
Please merge all topics about that its so simple... sorry for bothering :/

closed 2018-01-26 by nefarius