nefarius/ScpToolkit #661: Add Support for OPP Gamepad

Zero3K, 2017-03-09:
Its the name given to the newer model of the PowerA Pro Elite Wireless Controller. I am attaching the HID report and the debug information to this issue.

[ScpToolkit_Log-Package_Bryan_1489041504.zip](https://github.com/nefarius/ScpToolkit/files/829976/ScpToolkit_Log-Package_Bryan_1489041504.zip)
[OPP_hid-report.dump.txt](https://github.com/nefarius/ScpToolkit/files/829975/OPP_hid-report.dump.txt)

Zero3K, 2017-03-10:
Here is what the Application Event Log contains when I installed the WinUSB driver over the default one that is used by the controller:

[code]Application: ScpServer.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.ArgumentException
   at System.Buffer.BlockCopy(System.Array, Int32, System.Array, Int32, Int32)
   at ScpControl.Usb.Ds3.UsbDs3.Process(System.DateTime)
   at System.Reactive.AnonymousSafeObserver`1[[System.Int64, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].OnNext(Int64)
   at System.Reactive.Linq.ObservableImpl.Timer+TimerImpl.Tick(Int64)
   at System.Reactive.Concurrency.DefaultScheduler+<>c__DisplayClass9`1[[System.Int64, mscorlib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089]].<SchedulePeriodic>b__7()
   at System.Reactive.Concurrency.AsyncLock.Wait(System.Action)
   at System.Threading.ExecutionContext.RunInternal(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.ExecutionContext.Run(System.Threading.ExecutionContext, System.Threading.ContextCallback, System.Object, Boolean)
   at System.Threading.TimerQueueTimer.CallCallback()
   at System.Threading.TimerQueueTimer.Fire()
   at System.Threading.TimerQueue.FireNextTimers()[/code]

[code]Faulting application name: ScpServer.exe, version: 1.7.277.16103, time stamp: 0x570d1b03
Faulting module name: KERNELBASE.dll, version: 6.1.7601.23572, time stamp: 0x57fd0696
Exception code: 0xe0434352
Fault offset: 0x000000000001a06d
Faulting process id: 0xe60
Faulting application start time: 0x01d2993dcf06cd70
Faulting application path: C:\Program Files\Nefarius Software Solutions\ScpToolkit\ScpServer.exe
Faulting module path: C:\Windows\system32\KERNELBASE.dll
Report Id: 61036bc0-0531-11e7-af72-00dbdf7e9dc4[/code]

Zero3K, 2017-03-10:
Does anyone else have one?

Zero3K, 2017-03-10:
I added 00:00:00 as a Genuine MAC address to the ScpControl.ini. It doesn't crash anymore but now all the buttons except for the Square button don't work (the Square Button causes some of the other buttons to be pressed when its pressed). Also, according to the Scp Monitor tool, the packets don't increase unless I hold the Square button.

LucaSoranota, 2017-03-14:
ahh this is the same issues as mine evertime i press square the hitreg are going crazy lol and in 1,7 the app cant detect my controller at all

Zero3K, 2017-03-19:
This is the controller I am writing about (the only things different between the listing and the one I actually have is that it has a different grip, better joysticks, a different USB dongle, and that it requires AAA batteries):

https://www.amazon.com/Pro-Elite-Wireless-Controller-PS3/dp/B003V4AK8E/

Zero3K, 2017-05-02:
Any news regarding this issue?

Zero3K, 2017-06-26:
I guess it won't be supported.

Zero3K, 2017-12-10:
Well?

closed 2018-01-26 by nefarius