nefarius/ScpToolkit #663: SCP Toolkit refuses to work at all, DS3 Controller not working. WIN10

SakuraRico, 2017-03-10:
Hello everyone,
    Last week was the last time I managed to normally use my official PS3 Dualshock 3 controller on my PC. I believe after which the PC updated some stuff.
     Now when I boot up my PC, the controller tries to connect to it (I have it via USB, since I don't have bluetooth), and the lights keep on blinking until they just turn off eventually. I tried to reinstall my drivers, since when it does that, reinstalling is usually how I fix it, but when I try to go to the driver installer on the program, I get this.
[image]

It says to consult the logs, but I understand nothing that's written there. I tried using Better DS3 which I had as a backup, but since my controller doesn't seem to work with my PC anymore, it doesn't show up.

I was using the controller normally, I don't know where it went wrong. The SCP Monitor gives me this.

[image]

I am stumped as to what I should do. I can't play my games because keyboard is so uncomfortable. I included the logs, just in case it helps anyone decipher what's going on.

[Logs.zip](https://github.com/nefarius/ScpToolkit/files/835245/Logs.zip)

Any help?

Zero3K, 2017-03-11:
Did you try running the Clean Wipe program?

SakuraRico, 2017-03-11:
I did. Also tried uninstalling SCP Toolkit, and doing a fresh installation. Neither worked.

Kinsoku, 2017-03-17:
I'm having almost the same problem after the last update of WIN10. Try to manually open SCP Server and your PC should recognize your controller.

anacondadreams, 2017-03-18:
I have the same issue. Since the latest Win10 update BetterDS3 hasn't worked at all so I tried SCP. It could just be that they need time to get up to speed with the latest Win10 update.

SakuraRico, 2017-03-23:
I tried opening the SCP Server, it gave me some error messages and nothing happened.

closed 2018-01-26 by nefarius