nefarius/ScpToolkit #665: Bluetooth not working.

indexftw, 2017-03-11:
Hi.
If bluetooth stops working i.e. after a reboot/sleep/hibernation, I have to delete BT device from device manager, reboot, reinstall BT driver with zadig. I get it that scp service can't hook up with bt device after changing states of pc. A lot of time waste tbh.
Is there a non ghetto workaround with this?

charlesmilette, 2017-03-11:
#51

indexftw, 2017-03-11:
@charlesmilette 
To follow that page I had to update to the newest stable version which is shy of l1r1/l2r2 remap feature.
Also newest version doesn't fix anything. Could you elaborate where to look in #51? It says "Open" I'm assuming no one found a fix? I don't get it, really.

charlesmilette, 2017-03-11:
I was just referring this issue because we are aware of the problem, and we
already have an issue open for it. This bug really depends on the hardware.
If you have one of the compatible Bluetooth controller listed on the wiki,
you shouldn't have this bug.

indexftw, 2017-03-12:
I see. I found a few workarounds. Do issues bump up top if you comment on them? I'm sorry if this clogs up this git.
Thanks for coding. The app is brilliant.

closed 2018-01-26 by nefarius