nefarius/ScpToolkit #666: Driver Detect Genuine PS3 but didnt work

LucaSoranota, 2017-03-11:
im using hori gem pad 3 DS3 controller. the driver and installation are working smoothly and the ds3server detect it as genuine ds3 controller but the controller didnt work at all and everytime i press square the button register on the gamepad setting on windows are getting crazy... any solution?

LucaSoranota, 2017-03-12:
[PAD A_hid-report.dump.txt](https://github.com/nefarius/ScpToolkit/files/836253/PAD.A_hid-report.dump.txt)

that is the report from the gamepad analyzer..

Zero3K, 2017-03-12:
Oh, it seems like you're having the same issue as me. Except I had to manually add the MAC address and Hardware ID to ScpControl.ini in order for it to be detected as a Genuine DS3 controller.

LucaSoranota, 2017-03-12:
the thing is i even cant install the driver because on 1,7 my devices wont show.. i tried to find a fix and alternative about 4 hrs ago until now

closed 2018-01-26 by nefarius