nefarius/ScpToolkit #671: BSODs on W7 Pro

RogueLizard, 2017-03-16:
Getting SYSTEM_SERVICE_EXCEPTION with Bluetooth plugged in during startup. Uninstalled SCP driver, problem went away.

System specs:
AsRock H77M
i5 3330 at 3.8GHz
8GB Corsair XMS3
nVidia GTX 1050ti SC
nVidia GTX 460
600W Corsair CX600

Latest crash playing Fallout 4 on 3 screens (4280x1024)

rougetimelord, 2017-03-17:
lol, write your own tool if you want to be so disrespectful 😄

charlesmilette, 2017-03-17:
Works on my machine™

RogueLizard, 2017-03-17:
I'm gonna stick with my PS3 controller for now with Better DS3 and
MotionInJoy, something that I have NEVER had any issues with...take hint
Nefarious...

DriverVerifier hates the virtual bus driver, by the way. You should fix
that...lots of issues with it that produce blue screens.

deadair3210, 2017-03-17:
MotionJoy has been shown multiple times to cause more harm than good.
ScpToolkit works fine on my machine, as well as many many others, the fact that you seem to think that the dev needs to immediately jump to action just because it doesnt work on your machine makes my head hurt.

People like you insulting devs just because you can't do anything else productive with your time is why most projects end up dying. Either help find and fix the problem if you can, or put a issue report in and wait like everyone else. You have no right to try to rush someone without any intention of helping

charlesmilette, 2017-03-17:
In my case, MotionInJoy was a pile of trash that never worked. While with ScpToolkit it's literally plug and play.

It really depends on the hardware setup.

closed 2018-01-26 by nefarius