nefarius/ScpToolkit #679: SCP Settings Manager Unexpected Error?

nendraharyo, 2017-03-21:
Hi could anyone help me? I put another ScpControl.dll from thread #652 because my controller is fake and the original dll doesn't detect my controller wirelessly. After i put the new dll it's now functioning but when i tried to open SCP Settings Manager it gives me an error message as shown below. I put back the original dll and it works fine. Any ideas to solve this? Thanks :)
[image]

thaleslipk, 2017-03-29:
@nendraharyo Man, i dont have time now to inspect this problem on the code, so the solution i have so far for you is just replace this modded dll by the original dll and open the settings change what you want and after you finish stop the service replace it again by the modded dll for fake controllers and its done.

When i have more time i will look deeper on the code to see why settings doesnt open, and i will update the file on the post and put solution there, but this issue you can close.

nendraharyo, 2017-03-30:
Yup, currently doing that but to enable vibration in pcsx2 i need the settings manager to "inject" the driver and it "injects" as soon as the emulator starts. i closed this issue  and wait for your update in your thread. 

But seriously you're awesome for creating this dll even though it have some problem but i'm sure it'll be easy to fix if you have the time for it. Looking forward for the updates:)

closed 2017-03-30 by nendraharyo